LDMX-Software/ldmx-sw #966: Tab Completion when interacting with container

tomeichlersmith, 2021-03-01:
I just found out that you can [customize how your commands respond to tab-completion requests](https://iridakos.com/programming/2018/03/01/bash-programmable-completion-tutorial). This is pretty rad, and I want to do it. My idea is to have `ldmx` be **the** command with multiple sub-commands (kinda like how `docker` has multiple sub-commands). The command tree I'm thinking of implementing is

- `ldmx`
     - `list` : list tags for the input repo name
     - `clean` : remove all containers and container images on the local machine
     - `config` : print out configuration to stdout
     - `pull` : pull down input container image and set it to be used
     - `use` : use input container image and only pull down if it doesn't exist on local machine
     - `run` : Got to first argument and then run everything after it in the container (basically what the entry-point script does)
     - `fire <stuff>` : Equivalent to `run $(pwd) fire <stuff>`
     - `make <stuff>` : Equivalent to `run $(pwd) make <stuff>`
     - `cmake <stuff>` : Equivalent to `run $(pwd) cmake <stuff>`
     - (and so on with other ldmx-sw commands, I think I will be able to automate this to pick up all executables in the ldmx-sw install)

bryngemark, 2021-03-01:
cool! would `ldmx mount` be a good idea? primarily for when you want to run on data from some input path.

tomeichlersmith, 2021-03-02:
Great idea! I will look into it :+1:

tomeichlersmith, 2021-03-12:
`ldmx mount` will be included, I've opened up a draft PR #970 to point out my progress. I still need to check that it works on `singularity` systems.

PR #970 (2021-04-13): Iss966 - Update container-interacting bash functions
Since PR #965 requires some tweaks to the environment script, I have developed this updated container interface on top of those changes.

This resolves #966 by implementing what was discussed there, not much else to it.

At the request of @jmmans , I've included the **great idea** of having a `.ldmxrc` to persist the local configuration of the ldmx-sw computing environment. In the tree currently is a default configuration as an example.

:thinking: Maybe I want to write my own shell. :thinking: We could call it `ldmxsh` :laughing: 

[Why end [... cut]
closed 2021-04-13 by omar-moreno