LDMX-Software/ldmx-sw #968: Make the number of earlier/later bunches used for pileup independent

bryngemark, 2021-03-12:
In the current implementation, out-of-time pileup is placed symmetrically in +/- _m_ bunches. This should be separately configurable (so place pileup in a random bunch from a continuum starting 3 bunches before, to 10 bunches after, or whatever). 

**Describe the solution**
I'm going to change the setup in the `OverlayProducer` such that two numbers are specified instead of one.

GNiramay, 2021-03-12:
I was wondering, can we also specify a physical time range for the pileup? I mean, instead of setting +/- 4 bunches, can we specify +/- 100ns perhaps?

bryngemark, 2021-03-16:
Hi @GNiramay, sorry, my github notifications have been ending up in spam lately :(

I think we do want to operate under the assumption that the beam will be bunched? That said, it is already possible to smear the time within a bunch (applies also to out-of-time bunches). I've set this smearing to a Gaussian distribution, but that could be tweaked/configurable if there is a use case for it. Any thoughts from @awhitbeck?

PR #969 (2021-03-18): Some long-anticipated tweaks to the overlay producer (event wrapping and start event randomization)
This adapts `EventFile` as needed to have a special treatment in `nextEvent` when the file used is a pileup input file (used for overlay). All it does is to reset the event counter to start from the beginning, instead of return/crash, when we reach the end of the tree. 

In addition, the pileup event _start number_ is randomized (on an interval up to 500, a little arbitrary but I don't think we'll reuse the same pileup file more than at the very most ~100 times, and I couldn't figure out immedia [... cut]
closed 2021-03-18 by omar-moreno