LDMX-Software/ldmx-sw #985: HCal Detector Simulation

tomeichlersmith, 2021-05-04:
Currently, the HCal detector in our simulation differs drastically from when we expect outside of the simulation. Similar to the [ECal geometry](https://github.com/LDMX-Software/ldmx-sw/issues/890), realism improvements must be made. Personally, the only change that I know of is that the back HCal is planned to be 2m by 2m rather than the 3m by 3m currently in the simulation. I have no doubt there are other updates that need to be made as well.

omar-moreno, 2021-11-02:
The reason it's 3x3 m^2 in the simulation was because it allowed studying of smaller geometries by just restricting the number of layers used in analysis. At this point, I don't see a consensus as to what the size of the HCal is going to be so it's likely we will leave it as a 3x3 m^2 detector for the foreseeable future. 

Beyond the geometry changes, it seems like all other issues should go into the submodule instead?  If you agree, I say we close this for now.

tomeichlersmith, 2021-11-02:
yea, that makes sense.

closed 2021-11-02 by tomeichlersmith