LDMX-Software/ldmx-sw #996: Failure to Launch Display-Connected Programs in WSL

tomeichlersmith, 2021-06-08:
**Describe the bug**
ROOT is unable to detect the X-window server from within the container when running docker inside of Windoze's Subsystem for Linux (WSL). This issue seems to be a miscommunication between the host OS (ubuntu inside WSL) and the container.

**To Reproduce**
Steps to reproduce the behavior:
1. Open WSL with Ubuntu 20.04 installed on it
2. Initialize ldmx computing environment: `source ldmx-sw/scripts/ldmx-env.sh`
3. Attempt to launch root browser: `ldmx rootbrowse`
```bash
$ ldmx rootbrowse
Error in <TGClient::TGClient>: can't open display ":0", switching to batch mode...
In case you run from a remote ssh session, reconnect with ssh -Y
Press enter to exit.
```
The shell is still interactive - i.e. the user can still press <enter> and root does exit - so it is something specificaly missing about the display.

**Desired behavior**
We'd love for the display to be connected automatically within WSL as other OSs. We'd settle for a guide showing how to change WSL settings to allow it. We'd be grumpy if it flat-out won't work.

**Environment:**
 - Inside the container:
    - Output of `ldmx-container-config`
 - Outside the container:
    - Using WSL on Windoze
    - OS Installed in WSL:
```
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL=https://www.ubuntu.com/
SUPPORT_URL=https://help.ubuntu.com/
BUG_REPORT_URL=https://bugs.launchpad.net/ubuntu/
PRIVACY_POLICY_URL=https://www.ubuntu.com/legal/terms-and-policies/privacy-policy
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

**Additional context**
@D-Covey09 originally found this issue and since his computer has Windoze on it and mine doesn't, I will be working with him to try to resolve this issue.

tomeichlersmith, 2021-06-08:
## Research Notes
- Run X11 server _outside_ WSL: https://stackoverflow.com/q/64047328
- Stack Overflow Answer to try: https://superuser.com/a/1476160
- Running Desktop Apps with Docker (includes Windoze but maybe not WSL?): https://betterprogramming.pub/running-desktop-apps-in-docker-43a70a5265c4
- Looks like we can check the [OS environment variable to see if we are in WSL](https://github.com/sindresorhus/is-wsl/blob/main/index.js)

D-Covey09, 2021-06-09:
A decent solution appears to be running 
`source ldmx-sw/scripts/ldmx-env.sh
export LDMX_CONTAINER_DISPLAY=$(cat /etc/resolv.conf | grep nameserver | awk '{print $2}')`

running root in the ldmx container will now properly display a X window on windows. You can put these in your `.bashrc` file, just replace the path of ldmx-env.sh with the absolute path. 

A major issue for now seems to be that root / WSL seems to loose connection to the X server after a few minutes of inactivity. I'm unsure if it is on a timer regardless of activity, or if it's triggered by not interacting with the ubuntu bash shell for a time. The X server disconnects from the shell and WSL locks, becoming completely unresponsive, only echoing back my keystrokes. Necessitating a relog into WSL. I'll post a fix when I find it. 
Here's the error code accompanying that disconnect:
`Error in <RootX11IOErrorHandler>: fatal X11 error (connection to server lost?!)
**** Save data and exit application ****
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/deps/cernroot/lib/ROOT/_application.py", line 91, in _process_root_events
    gSystem.ProcessEvents()
SystemError: bool TSystem::ProcessEvents() =>
    SystemError: problem in C++; program state was reset`

PR #997 (2021-06-18): Integrate WSL into display
https://wiki.ubuntu.com/WSL#Running_Graphical_Applications for the details.

Essentially, we are assuming that an X server is running somewhere and we point our container directly to the IP of that server. On linux systems, the X server (probably) requires no setup while on Windoze, the user will need to install some X server for WSL to connect to. We didn't have this problem before because the `DISPLAY` variable on Linux systems is set to `:0` and some other nonsense can automagically connect to the X server. WSL can't do that.

I am updating _ldmx-sw_, here are the [... cut]
closed 2021-06-18 by tomeichlersmith