LDMX-Software/ldmx-sw #1018: Possible off by one error in buildStripPositionMap

EinarElen, 2021-08-20:
**Describe the bug**
Currently, the calculation of the layer center (in z) is calculated as 
``` 
// layer * (layer_dz) + (layer_dz)/2
double layercenter =
            layer * LayerThickness_.at(section) + 0.5 * ThicknessScint_;
```
However, the layer number is not zero-based and starts at 1. Therefore, the layer center of the first layer is more than one layerthickness away from the zeroLayer which doesn't seem right. To get the correct position for the prototype Hcal, this has to changed as in 7453a27. However, if I'm missing something and this is intended, we can work around it in the prototype setup

tomeichlersmith, 2021-08-20:
I don't think you are missing something. I think you are correct and we didn't notice because the HCal is so uniform and large, such a small change was not seen.

EinarElen, 2021-08-20:
Makes sense. The fix is already in the prototype geometry branch, so I guess there's no rush to fix it here?

tomeichlersmith, 2021-08-20:
No rush - unless other people weight in, I'm comfortable waiting until we merge in the prototype geometry in order to get this fix.

cmantill, 2021-08-20:
hi, I agree with waiting for the prototype geometry. Thanks @EinarElen for catching this!

EinarElen, 2021-11-02:
Hi @omar-moreno, what does this assignment represent? This was resolved when we merged 7453a27

omar-moreno, 2021-11-02:
It's for bookkeeping purposes only. You can ignore it.