LDMX-Software/ldmx-sw #1021: Compatibility for Older Singularity Installs

tomeichlersmith, 2021-09-13:
**Describe the bug**
Older versions of singularity (most [notably version 3.5.2](https://sylabs.io/guides/3.5/user-guide/cli/singularity_run.html) installed at [UCSB](http://csc.cnsi.ucsb.edu/docs/pod-cluster)) does not have the `--env` parameter allowing us to define a select set of environment variables to be defined for inside the container.

**Desired behavior**
It would be nice for the environment script to detect the singularity version and then define the correct wrapping function.

**Additional context**
@davidgjiang detected this bug during his work developing on the UCSB cluster.

tomeichlersmith, 2021-09-13:
We can wrap our singularity command with a call to [env](https://man7.org/linux/man-pages/man1/env.1.html) making sure only the environment variables we want persisted are persisted i.e.

```bash
# in _ldmx_run for the singularity branch
env -i LDMX_BASE=${LDMX_BASE} csv_list=${csv_list} LDMX_SINGULARITY_IMG=${LDMX_SINGULARITY_IMG} \
  singularity run --no-home --bind ${csv_list} ${LDMX_SINGULARITY_IMG} "$@"
```

Probably further modification will be necessary.

tomeichlersmith, 2021-12-08:
@davidgjiang @p-masterson 

Are either of you able to comment on the singularity version at UCSB? I'd really like to avoid adding in this level of complexity to the environment script if you are able to update singularity there. Obviously if no path to updating is possible, then I can investigate this issue further.

p-masterson, 2021-12-08:
Hi Tom!  I think the default version at pod is 3.5.2, and 2.4, 2.4.6, and
2.6 are also available without any complications.  Are we going to need a
later/earlier version for future ldmx-sw updates?

tomeichlersmith, 2021-12-08:
The issue is that the environment script assumes that `singularity run` has the `--env` parameter for sharing specific environment variables with the container. If you aren't using the environment script at pod and just using it for large scale purposes, then we are all set.

I cam across this issue when trying to help David get started. I have since lost track and so I wanted to ask you both where we are at.

p-masterson, 2021-12-10:
Ah, I remember that one...we're using the container at pod and will
probably keep doing so, but I think we found in the past that commenting
the env line would get rid of any issues.  Is that still going to be a
consistent solution, or do we definitely need to include the line in future
versions?

tomeichlersmith, 2021-12-10:
Good question. It is definitely a good solution if it works at your site. The reason we (often) want to use `--cleanenv` and `--env` is so that we don't accidentally "pick up" any dependencies that exist outside of the container.

Specifically, I implemented the `env` flags because at UMN, there was a `ROOTSYS` environment variable _outside_ the container which would conflict with the `ROOTSYS` _inside_ the container leading to cryptic errors. I have since updated the container image to be more robust against these issues, so just commenting out that line should be safe.

I'm just going to close this issue because I think we have a working solution.