LDMX-Software/ldmx-sw #1037: Apply HCal Unpacking Knowledge to ECal

tomeichlersmith, 2021-11-15:
We did a lot of quick development of the HCal unpacking pipeline during the testbeam. I am planning on copying a lot of what we learned back over to the ECal. Since we weren't able to fully test the unpacking pipeline, the work that will end up being merged in will not be 100% ready; however, it will be significantly closer than what we currently have.

## Locations
- DetDescr/EcalElectronicsID (learn from Hcal EID)
- Ecal/EcalRawDecoder (learn from Hcal/HcalRawDecoder)

tomeichlersmith, 2021-11-18:
@jmmans I need some insight on how to proceed.

For the testbeam, we were orienting our decoding to work with a _single_ FPGA since that was all that was needed to readout our small prototype HCal (to my knowledge). Even if in reality this is not correct, the software decoding the event buffer has a few places where a single FPGA is assumed.

Anyways, the basic structure in the software follows the data format outlined in your ECal DAQ specification document with
an event packet holding all the sample packets for all the links connected to an FPGA. Does this scale to multiple FPGAs? i.e. Will each FPGA wrap all the sample packets for all the links connected to it or will there be some other, higher level, subsystem event building after the FPGAs?

jmmans, 2021-11-18:
A few comments, some which are a bit random:

 (1) I generally refer to the FPGA on the front-end specifically as the Polarfire for clarity.  The data will pass through multiple FPGAs in its processing, so it can be ambigous what a question is asking.

 (2) I expect the Polarfire will continue to work as it does now -- by handling single readout requests, even if multiple readout requests are needed for a single event.  

 (3) The DPM FPGA/uMNio FPGA (let's call it the backend-FPGA), is where the bundling of readout requests into an event occurs.  In the uMNio right now, this happens in software, while for the DPM I've drafted firmware which does the last.  Firmware would be long-term goal.

 (4) The DPM will hand its data to the central DAQ through the AXI-S interface.  At that point, I don't know what happens.  In CMS, each different DPM would show up in the system as an independent top-level data block.  I don't know how LDMX CDAQ plans for things to work -- whether there will be some sort of by-subdetector wrapping which is done.  I would argue that a separation by subdetector doesn't really seem necessary -- better to just make a container with many labelled blocks ('ECal-21','TS-01','Tk-04') and the unpacker grabs out all the appropriate blocks and unpacks them to create the digi collection.

 (4) To this point, we have "one optical link per backend-FPGA" or "one optical link per raw-block", so there is the last question you raise -> how is the data handled when multiple Polarfires feed a single backend-FPGA.  I expect the same sort of formatting structure would be used, just one level up.  However, it could be done 'link-by-link' so that the full event from a given link is given followed by the full event from the next link, or 'sample-by-sample' so that all of the first samples from all links come first, then the second samples from all links and so on.  If there is no reason to do 'sample-by-sample', I'd expect a link-by-link structure.  We could add a header already now to represent this missing level of structure.

 (5) We may want to eventually consider an even-more-compact storage of ROCs which have no samples that pass ZS.  Indeed, for archival purposes, we may need a different raw data format which removes a lot of the headers and trailers.  Otherwise, we'll blow up our data format with accounting information which we don't need.

tomeichlersmith, 2021-11-18:
Thank you for this elaboration, this makes sense.

PR #1038 (2021-11-30): Merge compiling draft of ECal raw {de,en}coders
Not fully operational but compiling and a good starting point

## Related Submodle PRs
- LDMX-Software/Ecal/pull/37
- LDMX-Software/Hcal/pull/19

This PR resolves #1037

Jeremy made some good comments on the linked issue about how the DAQ flow works. The current de/en coders are a good (compiling) place to wait for future decisions to be made by lower-level (i.e. closer to the detector) DAQ software.
closed 2021-11-30 by tomeichlersmith