LDMX-Software/ldmx-sw #1047: The HcalGeometry (in DetDescr) __str__ function doesn't work

EinarElen, 2022-01-15:
**Describe the bug**
Just some typos that need to be patched. 

**To Reproduce**
Try printing a HcalReadoutGeometry object :) 

**Desired behavior**
Output of all the parameters is printed without any errors.

EinarElen, 2022-01-15:
I went ahead and restructured the function as a whole to be a bit more useful for debugging things (and handling the difference between the three geometry types to some degree). Given that the function was broken, I'm going to assume that its safe to say that nobody was using it at the moment. 

Something along the lines of (from 9f2e2e6d4a48e6f3111936f445c9e1320719da3f)
```python
def __str__(self) :                                             
    """Stringify this configuration class"""                    
    s = '''                                                     
    HcalReadoutGeometry {{                                      
        Scintillator thickness: {:.1f} [mm], width: {:.1f} [mm],
        Number of sections: {{{}}},                             
        Layers: {{                                              
            Number of layers: {}                                
            Layer thickness: {} [mm]                            
            Z-position of zero-th layer: {} [mm]                
            Half total width of layers: {} [mm]                 
            Number of strips per layer: {}                      
            Location of zero-th strip per layer: {} [mm]        
        }},                                                     
        Ecal DX, DY: {}, {} [mm],                               
        Valid detector regexps: {}                              
    }}                                                          
    '''.format(self.ThicknessScint, self.WidthScint,            
            self.NumSections,                                   
            self.NumLayers, self.LayerThickness, self.ZeroLayer,
            self.HalfTotalWidth,self.NumStrips, self.ZeroStrip, 
            self.EcalDx, self.EcalDy,                           
            self.detectors_valid)                               
    return s                                                    
```

gives the following output for the different geometries
Prototype 
```
        HcalReadoutGeometry {
            Scintillator thickness: 20.0 [mm], width: 50.0 [mm],
            Number of sections: {1},
            Layers: {
                Number of layers: [19]
                Layer thickness: [49.0] [mm]
                Z-position of zero-th layer: [-438.5] [mm]
                Half total width of layers: [200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0] [mm]
                Number of strips per layer: [8, 8, 8, 8, 8, 8, 8, 8, 8, 12, 12, 12, 12, 12, 12, 12, 12, 12, 12]
                Location of zero-th strip per layer: [200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 200.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0, 300.0] [mm]
            },
            Ecal DX, DY: 0.0, 0.0 [mm],
            Valid detector regexps: ['ldmx-hcal-prototype-v1.0', 'ldmx-hcal-prototype-v1.0[.].*']
        }
```
v12
``` 
        HcalReadoutGeometry {
            Scintillator thickness: 20.0 [mm], width: 50.0 [mm],
            Number of sections: {5},
            Layers: {
                Number of layers: [100, 28, 28, 26, 26]
                Layer thickness: [49.0, 44.0, 44.0, 44.0, 44.0] [mm]
                Z-position of zero-th layer: [820.0, 300.0, 300.0, 300.0, 300.0] [mm]
                Half total width of layers: [1550.0, 972.0, 972.0, 916.0, 916.0] [mm]
                Number of strips per layer: [62, 12, 12, 12, 12]
                Location of zero-th strip per layer: [1550.0, 220.0, 220.0, 220.0, 220.0] [mm]
            },
            Ecal DX, DY: 800.0, 600.0 [mm],
            Valid detector regexps: ['ldmx-det-v12', 'ldmx-det-v12[.].*', 'ldmx-det-v9', 'ldmx-det-v10', 'ldmx-det-v11']
        }
```        
v13
```
        HcalReadoutGeometry {
            Scintillator thickness: 20.0 [mm], width: 50.0 [mm],
            Number of sections: {5},
            Layers: {
                Number of layers: [100, 28, 28, 26, 26]
                Layer thickness: [49.0, 44.0, 44.0, 44.0, 44.0] [mm]
                Z-position of zero-th layer: [820.0, 300.0, 300.0, 300.0, 300.0] [mm]
                Half total width of layers: [1550.0, 972.0, 972.0, 916.0, 916.0] [mm]
                Number of strips per layer: [62, 12, 12, 12, 12]
                Location of zero-th strip per layer: [1550.0, 220.0, 220.0, 220.0, 220.0] [mm]
            },
            Ecal DX, DY: 800.0, 600.0 [mm],
            Valid detector regexps: ['ldmx-det-v13', 'ldmx-det-v12', 'ldmx-det-v12[.].*', 'ldmx-det-v9', 'ldmx-det-v10', 'ldmx-det-v11']
        }
```

Which seems reasonably clear to me. Some lines get a bit long but fixing that seems like it would be more effort than it is worth

tomeichlersmith, 2022-07-20:
Fix implemented by PR mentioned above.

closed 2022-07-20 by tomeichlersmith