LDMX-Software/ldmx-sw #1053: Patch to Electronics ID deduction

tomeichlersmith, 2022-03-25:


jmmans, 2022-03-25:
Sorry, what?  If a link goes down, for example, it's down.  For unpacking purposes, the two half chips are independent.  I disagree with this proposal.

jmmans, 2022-03-25:
Also, if this is related to HCAL, HCAL doesn't use several of the channels /anyway/ because it operates in sum-by-four mode for trigger primitives.  I would NOT redesign the software id on this basis.

tomeichlersmith, 2022-03-25:
Where would the translation between the channel numbers listed physically on the board and the channel numbers as given in the independent links be done? Does the detector mapping need to already include this translation? Does the HcalDetectorMap loader perform this translation while reading?

PR #1054 (2022-03-28): patch to electronics id deduction
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1053 

### Related Sub-Module PRs

- https://github.com/LDMX-Software/Hcal/pull/26
closed 2022-03-28 by tomeichlersmith