LDMX-Software/ldmx-sw #1067: Testbeam reconstruction wrong barID assignment

PeterGy, 2022-05-10:
**Describe the bug**
During reconstruction, each pair of bars is assigned the same ID, one of them being correct and the other wrong.

**To Reproduce**
Steps to reproduce the behavior:
1. Decode a testbeam run, in this case run 287 fpga0, using decode.py found in [ldmx-tb-online](https://github.com/LDMX-Software/ldmx-tb-online/blob/main/near-offline/hcal/decode.py)
2. List every reconstructed barID in an event. You will find that half are missing while the others are duplicates.

**Desired behavior**
Bars recieving their correct IDs.

**Screenshots**
One image shows the mapping of hits.
[image]
Here is a full list of the present reco and digi IDs in an event. The true or false refers to whether the recoID translated from the digiID is in the event or not.
[Debug.txt](https://github.com/LDMX-Software/ldmx-sw/files/8662690/Debug.txt) 
As produced by the code snippet
[image]

**Environment:**
 - Inside the container:
    - Reconstruction performed on latest ldmx-sw with main container
    - Analysis performed on latest ldmx-sw `ldmx use dev iss41-boost-python` and the `iss1048-detectorID-bindings`  branch
 - Outside the container:
    - Reconstruction performed on Lunarc
    - Analysis performed on WSL

**Additional context**
Clearly ldmx-sw correctly converts digiIDs to recoIDs in general, but fails to do so somewhere along the testbeam reco.
The simulation reco works as intended.

cmantill, 2022-05-16:
Was able to reproduce this issue. 
It is a bug on the HcalRecProducer, fixed with [this commit](https://github.com/LDMX-Software/Hcal/commit/5c33cfaab9c26913ef48b9ec06f5c0fe6dcd3ab9). 
Should be integrated when issue 40 is fully tested.

tvami, 2024-09-17:
@cmantill given your commit, is this issue fixed and ready to be closed?

cmantill, 2024-09-18:
It looks like this branch was never integrated. But the reconstruction got split into two different files: https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Hcal/src/Hcal/HcalDoubleEndRecProducer.cxx
https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Hcal/src/Hcal/HcalSingleEndRecProducer.cxx

and the first file contains this fix. I am not sure if this needs to be propagated to HcalRecProducer.