LDMX-Software/ldmx-sw #1085: trigger OR processor, needed for multielectron

bryngemark, 2022-09-08:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to OR (or AND) triggers and then trigger skim on that, for future trigger efficiency studies. 

**Describe the solution you'd like**
I think the easiest is to make a new processor, which takes a configurable list of trigger collections, a configurable logic (doOR, doAND), and given that, checks whether the event passes the trigger and satisfies the desired logic, and sets the storage hint for the event to keep or not. Then this processor is what is used to set the skim decision, not the individual ones. The individual trigger decision can still be seen in each trigger collection (`pass_`).

**Describe alternatives you've considered**
I've been thinking that the storage hint/skimming rules should change but I think it's a larger rework than I'm happy to do. A processor seems versatile enough for me. 

This should absolutely go in before the next production release so we can get going with trigger skimming the multielectron samples we already have.

mrsolt, 2022-09-08:
I support this. I was also wondering how to get a skim for the missing energy visible A' trigger for the large scale production, and I think this could be useful. The trigger logic for this visibles trigger is already in the software, and we validated it. You just select the front portion of the Ecal to sum the energy for a missing energy trigger, instead of the entire Ecal.

Unfortunately, the standard missing energy trigger is a subset of what would be the visible trigger skim, so these can't be derived directly from the standard trigger. Thus, some sort of multi-trigger framework would be very useful.

bryngemark, 2022-09-09:
Great! Yes my thought would be that this processor is just checking which of the individual specified triggers pass the event and then selects what to do based on OR/AND settings, so the actual specific trigger logic is kept separate. (Clarifying the idea here for @rodwyer100 who has volunteered to actually code it up.)

rodwyer100, 2022-10-10:
[recoSkim3.txt](https://github.com/LDMX-Software/ldmx-sw/files/9749340/recoSkim3.txt)
[recoSkim2.txt](https://github.com/LDMX-Software/ldmx-sw/files/9749341/recoSkim2.txt)

rodwyer100, 2022-10-10:
Adding validation code in comment as per LK's desires.

rodwyer100, 2022-10-12:
Here is the image generated with the other validation architecture as requested in LK's last message

[image]

These are the scripts used to run the code:

[recoSkim5.txt](https://github.com/LDMX-Software/ldmx-sw/files/9768915/recoSkim5.txt)
[recoSkim4.txt](https://github.com/LDMX-Software/ldmx-sw/files/9768916/recoSkim4.txt)

rodwyer100, 2022-10-12:
Modifications of the validation scripts as per suggestions in pull request

[recoSkim3.txt](https://github.com/LDMX-Software/ldmx-sw/files/9769155/recoSkim3.txt)
[recoSkim4.txt](https://github.com/LDMX-Software/ldmx-sw/files/9769156/recoSkim4.txt)
[recoSkim5.txt](https://github.com/LDMX-Software/ldmx-sw/files/9769157/recoSkim5.txt)
[recoSkim2.txt](https://github.com/LDMX-Software/ldmx-sw/files/9769158/recoSkim2.txt)

closed 2022-10-14 by tomeichlersmith