LDMX-Software/ldmx-sw #1086: add a v14lyso branch

joyang8caltech, 2022-09-13:
**Describe alternatives you've considered**
Add a new v14lyso branch that contains geometric descriptions of active LYSO target in constants.gdml and target.gdml

awhitbeck, 2022-10-21:
Thank @joyang8caltech for getting this into github.  You'll need to integrate this with the rest of LDMX.  Currently, you have removed the rest of the detector.  I would start from the detector as it is in #1107 .  It is important that you check that a rendering of the target does not interfere with the tracking detectors.  

You have also changed the plastic detectors.  I suggest we leave these as they are in [target.gdml](https://github.com/LDMX-Software/ldmx-sw/blob/iss1102/Detectors/data/ldmx-det-v14/target.gdml).  These need to be consistent with the upstream modules otherwise we risk having subtle issue show up in the reconstruction.  It will be easier to maintain everything if the syntax of the plastic modules are all the same.  

Once you have this integrated with the rest of LDMX detector system, I would start testing in simulation.  You can use this configuration file: 

```
import math
from LDMX.Framework import ldmxcfg
from LDMX.SimCore import sim_example
p = sim_example.process()
p.sequence[0].setDetector("ldmx-det-v14")
p.sequence[0].verbosity = 10
p.termLogLevel = 0
import LDMX.Ecal.EcalGeometry
import LDMX.Hcal.HcalGeometry

p.maxEvents = 100
```

You can run this with: `ldmx fire test.py`, where `test.py` is the name of the file you drop this configuration into.  You will get a root file with all of the simhit collections for all of the subsystems saved.  

This will fire electrons through all of the TS and target modules and save the sim hits.  Checking the simhit collections are saved and that their distributions are reasonable for this pencil beam is critical.

joyang8caltech, 2022-10-24:
Hi Andrew:

Thanks for your suggestions.

 Plastic scintillator was changed in my targetW.gdml and targetLYSO.gdml because Geant4 gave errors to the original codes and didn't show the target.gdml.  I'll try to update the codes in #1107 as you suggested.

I will also download the rest of the detector and check if my new target doesn't interfere with the tracking detectors.

Thanks again.

Best regards,
James

tvami, 2024-06-07:
I think this is a duplicate issue of https://github.com/LDMX-Software/ldmx-sw/issues/1226

closed 2024-06-07 by tvami