LDMX-Software/ldmx-sw #1087: move TS modules upstream of tagger tracker

awhitbeck, 2022-09-21:
For upcoming MC campaign, we would like to move the "down" TS module upstream of the tagger tracker.  

We will pull the TS modules out of the tagger volume and put them into a separate volume.  

Currently, we have a new description of the geometry in the V14 detector folder on `TS_geom_edits` branch of LDMX-SW and the SimCore module.

awhitbeck, 2022-09-21:
@bryngemark I have made a dedicated issue to moving the TS modules.  I pushed all of my code.  Thanks to help from @tomeichlersmith and @omar-moreno the new GDML is properly getting loaded by LDMX-SW and I can run simulations.  However, I don't see any sim hits in the output root file.  

Some caveats to where we are currently:

- currently I have commented out the target GDML, though I think this didn't get pushed to github.  If you run as the code is there, you might get complaints about overlaps.  I have ignored these in the hops that I can get everything working with the upstream modules first.  
- The upstream modules are at a first guess of the best location, but not optimized for the beam parameters we'll want to use.  I suspect the lack of sim hits means that the beam is not hitting the modules, but moving the beam starting position back didn't help.  @tomeichlersmith  can your visualization code accommodate tracks from a simple particle gun?

tomeichlersmith, 2022-09-21:
> can your visualization code accommodate tracks from a simple particle gun?

I have no idea, I haven't tried.

> I don't see any sim hits in the output root file.

Remember: The SDs are attached via python configuration now rather than in the GDML. This relies on the configuration using the correct names. https://github.com/LDMX-Software/SimCore/blob/0656170b61f39dd972c6b8fd01fd9707391001d5/python/sensitive_detectors.py#L105-L142

bryngemark, 2022-09-22:
hi @tomeichlersmith, i've been trying to get the simhits in place by adjusting the SD info. i think i can conclude that this is indeed the issue since i don't see any TS SDs being loaded when i run:
```
---- LDMXSW: Configuration load complete  --------
---- LDMXSW: Starting event processing --------
[ DetectorConstruction ] : Attaching Tagger_TrackerSD to LDMXTaggerModuleVolume_component0Sensor0
[ DetectorConstruction ] : Attaching Recoil_TrackerSD to LDMXRecoilL14ModuleVolume_component0Sensor0
[ DetectorConstruction ] : Attaching Recoil_TrackerSD to LDMXRecoilL56ModuleVolume_component0Sensor0
[ DetectorConstruction ] : Attaching hcal_sd to back_hcal_ScintBox_volume
[ DetectorConstruction ] : Attaching hcal_sd to sideTB_hcal_ScintBox_volume
[ DetectorConstruction ] : Attaching hcal_sd to sideLR_hcal_ScintBox_volume
[ DetectorConstruction ] : Attaching ecal_sd to Si_volume
 [ Process ] 1 : RunHeader { run: 1, detectorName: ldmx-det-v12, description: Upstream 4 GeV 1-electron beam
```
-- i _think_ the "v12" in the runheader description is just hardwired from somewhere; i'm using v14 (and i know it bcs i commented the target as andrew recommended, and it's not showing up either)

here's what i do to tweak it 

``` python
from LDMX.SimCore.sensitive_detectors import TrigScintSD
                                                                            
tsTagA=TrigScintSD(1, "TaggerA", "trigger_pad_taggerA_bar_volume")
tsTagB=TrigScintSD(3, "TaggerB", "trigger_pad_taggerB_bar_volume")
```
and i think this matches the SDs @awhitbeck has been setting up [here](https://github.com/LDMX-Software/ldmx-sw/blob/abbc79a1dd19522a8c50b3c342e4746250d4477b/Detectors/data/ldmx-det-v14/trig_scint.gdml#L44-L58)

any thoughts? let me know if you want a config to try it out.

awhitbeck, 2022-09-22:
right, if you change these lines in `SimCore/python/sensitive_detectors.py` you can see the the SDs get attached.

tomeichlersmith, 2022-09-22:
After creating the sensitive detectors, you need to give them to the simulator so that they are used.
```python
sim.sensitive_detectors = [ ... ]
```
The default connection is done in [simulator.setDetector](https://github.com/LDMX-Software/SimCore/blob/0656170b61f39dd972c6b8fd01fd9707391001d5/python/simulator.py.in#L90-L127) and uses the helper functions in `sensitive_detectors.py`.

BTW, the detector name in the run header description is parsed from [GDML](https://github.com/LDMX-Software/ldmx-sw/blob/abbc79a1dd19522a8c50b3c342e4746250d4477b/Detectors/data/ldmx-det-v14/detector.gdml#L158-L162) and the v14 detector just hasn't been updated on your branch yet.

awhitbeck, 2022-09-22:
thanks, @tomeichlersmith.  I think these function calls don't need to b changed.  I didn't change the name of these functions in my edits to SimCore [here](https://github.com/LDMX-Software/SimCore/blob/TS_geom_edits/python/sensitive_detectors.py#L120-L130).  I'll try to double check though.

awhitbeck, 2022-09-22:
ah, one more thing, is the detector name important within ldmx-sw?  Could the name being 'v12' cause issues?

tomeichlersmith, 2022-09-22:
yes, many geometry conditions rely on the detector name to know which geometry to load

tomeichlersmith, 2022-09-22:
in your case, since you are based on a GDML which is using v12 hcal and v12 ecal, it might be best to leave it as v12

awhitbeck, 2022-09-22:
I just checked that the TS SDs are in the list, but then I thought, maybe the name of these things might be important!  If I change the first string in these TrigScintSD call, I get sim hit collections!   So, now i am trying to understand what these magic lines do!  Where are these functions ('TrigScinSD()`) I am calling in 'sensitive_detectors.py' defined?  It seems like if I want to change the name of these things, I also need to change something else.

tomeichlersmith, 2022-09-22:
The function `TrigScintSD(...)` is the class constructor which calls the `__init__` method.

https://github.com/LDMX-Software/SimCore/blob/0656170b61f39dd972c6b8fd01fd9707391001d5/python/sensitive_detectors.py#L121-L142

awhitbeck, 2022-09-22:
well, I think I lied, I still don't see any hits, but I'll dig more and follow up with more information, hopefully.

awhitbeck, 2022-09-23:
okay, I got the sim hits.  The name discussion above was a red herring.  Now, the upstream modules are in places and working properly.  I still need to check things with something more complicated than a pencil beam and fix the target area module...

bryngemark, 2022-09-23:
ok. i will mention here for the record that the scoring planes haven't been updated yet either. not the first priority but something to keep on the list

PR #1101 (2022-10-17): TS DR Geometry Updates
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1087 

## Check List
- [ ] I successfully compiled _ldmx-sw_ with my developments

- [ ] I ran my developments and the following shows that they are successful.

_< put plots or some other proof that your developments work >_

- [x] I attached any sub-module related changes to this PR