LDMX-Software/ldmx-sw #1089: shorthand to pass environment variables to container

bryngemark, 2022-09-23:
**Is your feature request related to a problem? Please describe.**
There are use cases for passing an environment variable to the container. Examples include Geant4 specific stuff. 

**Describe the solution you'd like**
This is easy with docker: use `-e`. Right now this would mean tweaking the definition of `ldmx_run`. I'd like to be able to add variables just like I can add directories to mount and have that list passed under the hood when I run e.g. `ldmx fire`. I propose a new command `ldmx env` (or some other name) that works just like `ldmx mount` but for environment variables. 

**Describe alternatives you've considered**
Hacking `ldmx-env.sh` or using the entire docker run command explicitly. Not that great

tomeichlersmith, 2022-10-03:
I like this solution and have been thinking about it for some time. Is this something you are interested in implementing? If not, I can work on it as well.

bryngemark, 2022-10-10:
I'd be happy to give it a go! I'll look to `ldmx mount` for inspiration.

PR #1095 (2022-10-12): Set environment variables for container/image using function in ldmx-env.sh
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This adds a function to pass environment variables to a docker container or singularity image, using a function `ldmx setenv` which is set up by sourcing `ldmx-env.sh` as usual. This resolves #1089.

## Check List
- [ ] I successfully compiled _ldmx-sw_ with my developments -- N/A, this is not compiled code 

- [x] I ran my developments and the following shows that they are successful.

Environment variables get added to a bash [... cut]
closed 2022-10-12 by tomeichlersmith