LDMX-Software/ldmx-sw #1091: no ecal, hcal conditions for v14

bryngemark, 2022-09-25:
**Describe the bug**
This is a minor thing but as long as we don't have named ecal and hcal conditions for `v14` we won't be able to use that name in the updated geometry. 

**To Reproduce**
in Detectors/data/ldmx-det-v14/detector.gdml, replace 12 by 14 here:
``` xml
       <auxiliary auxtype="DetectorVersion" auxvalue="12">
       <auxiliary auxtype="DetectorName" auxvalue="ldmx-det-v12"/>
```
and you'll see
```
 [ fire ] 4 : [GeometryException] : Unable to create EcalHexReadout
  at /Users/lenebryngemark/work/LDMX/code/ldmx-sw/Ecal/src/Ecal/EcalGeometryProvider.cxx:137 in getCondition
```

**Environment:**
this is in the context of the `TS_geom_edits` branch

tomeichlersmith, 2022-10-03:
Apologies for the delay, I must have missed the creation of this issue.

There are v14 geometry conditions on `trunk` so the simplest (although maybe not easiest) route is to rebase your developments back onto trunk. The conditions on trunk contain the updated Ecal geometry as well as a slightly updated Hcal to accomodate the ECal changes.

https://github.com/LDMX-Software/ldmx-sw/blob/a4319646e3dce9cdca23147294d7775a20ef4493/DetDescr/python/EcalGeometry.py#L110-L125

And actually, right now, the HcalGeometry condition is just the same as v12 since we are still waiting for the actual Hcal v14 developments.

https://github.com/LDMX-Software/ldmx-sw/blob/a4319646e3dce9cdca23147294d7775a20ef4493/DetDescr/python/HcalGeometry.py#L92

bryngemark, 2022-10-03:
Excellent, thanks @tomeichlersmith! I'll go ahead and close this.

closed 2022-10-03 by bryngemark