LDMX-Software/ldmx-sw #1096: Update Hcal HGCROC Pulse Shape in Simulation

mrsolt, 2022-10-11:
The pulse shape in the HGCROC emulator for the Hcal is currently unrealistic. It does not match the current test beam data, and this could be a major factor for (early) data/MC mismatch, especially for energy deposition. For example, it both ramps up and down too quickly (and peaks in the first sample), while data has a slower ramp up (peaking in the ~4th sample) and does not fully ramp down in the 8 sample time window. Based on how we calibrate the TB data, this is creating issues for data/MC comparison.

We will implement a more realistic pulse shape. I already have an empirically derived working pulse shape that I have shown in a few meetings, but I am open to other ideas for more standard pulse shapes. This will improve not only our energy deposition, but also our TOA and TOT comparisons between data and MC.

jmmans, 2022-10-12:
While it is true that the shape is slower, tuning to testbeam may be
pessimistic because there are quite a few ROC parameters which could affect
this which we didn't test.

Also, remember the beam was asynchronous to the clock, so there is major
smearing from that.

mrsolt, 2022-10-12:
Thanks for the feedback Jeremy. We discussed this in detail in the test beam meeting today.

We decided it would be better to take a few simpler steps than to actually change the pulse shape. We loosely converged on the following steps.

1. Explicitly show the data/MC discrepancy to get an idea of the scale of the issues (we are close to this, but weren't ready to show it today.) Then show the digitization in both data/MC.
2. If it looks like digitization could be a large part of the discrepancy, try a few simple things to start (in this order):
a. Shift the waveform to peak where the data actually is.
b. Smear the timing of the pulse, since as you say the the beam was asynchronous to the clock.
c. Tune some of the parameters of the current pulse shape to better match data.