LDMX-Software/ldmx-sw #1102: TrigScint naming updates

tomeichlersmith, 2022-10-17:
**Describe the bug**
In the simulation module, the updated TrigScint geometry renamed the collections that it produced. This is a logical update which unfortunately has the cascading effect of requiring updates to the downstream input/output collection names.

In order to get the validation samples running, I commented out the TrigScint digi pipelines and the TrigScint DQM. This is obviously not long term, but I need someone more familiar with the TS pipeline to do the cascading name updates. Basically, I would like some updates allowing us to undo everything in https://github.com/LDMX-Software/ldmx-sw/commit/faad25ceaa5575a7b2c12ff587a3684190e84b16 .

@awhitbeck @bryngemark Maybe one of you has time this week to do this update?

bryngemark, 2022-10-17:
hi @tomeichlersmith, it's the default tag/up/down naming scheme you're talking about, right, or is there more in the validation workflow?

tomeichlersmith, 2022-10-17:
Yea - I am pretty sure its

Tagger -> TaggerA
Down -> TaggerB
Up -> Up

But I just need y'all to make sure its that simple.

bryngemark, 2022-10-17:
alright. i can take a look and push an update, if not today then definitely by then end of the week

bryngemark, 2022-10-17:
Haha oh, no you are right Tom, it's Down-->TaggerB, which comes *before* TaggerA. Did I mention I find it confusing..?

tomeichlersmith, 2022-10-20:
@awhitbeck noticed that the z position of the target sim hits are at `-845` which is _not_ near 0. Looking for a _longitudinal_ shift placing the TrigScint volumes upstream of the beam starting vertex.

awhitbeck, 2022-10-20:
There were some conflicts in the naming convention that was causing some weird behaviors.  I have fixed them in branch iss1102.  I checked that there are hit collection for all three of the TS modules and the target.  I have all retuned the beam parameters so that the beam goes through the center of all TS modules and the target.   Plots below show how close I got. 

[image]
[image]
[image]
[image]

beam parameters are: 

```p.sequence[0].generators[0].position=[-44.,0.,-880.]
theta = math.radians(5.65)
p.sequence[0].generators[0].direction=[math.sin(theta),0.,math.cos(theta)]
p.sequence[0].generators[0].energy=4.```

It would be helpful if someone could verify this is reproducible when they check out my branch.  

We just need to decide on a new naming convention.

bryngemark, 2022-10-20:
thanks @awhitbeck. if you start a PR, either of us can commit the name changes. i can do it after trying running sim+digi with your branch to verify, as you requested.

@tomeichlersmith, @omar-moreno, is the angle change (4.5 --> 5.65), needed when we move the beam starting point back, already included in default generator examples?

tomeichlersmith, 2022-10-21:
The angle change is not in SimCore, I will make a patch release of SimCore for that reason.

awhitbeck, 2022-10-21:
@tomeichlersmith I referenced an update to `SimCore` in my PR (#1107).  I also missed the position of the beam in my comment above; the angle and the starting position are critical.  Let me copy that information here, just to be sure: 

```
p.sequence[0].generators[0].position=[-44.,0.,-880.]
theta = math.radians(5.65)
p.sequence[0].generators[0].direction=[math.sin(theta),0.,math.cos(theta)]
p.sequence[0].generators[0].energy=4.
```

closed 2022-10-31 by tomeichlersmith