LDMX-Software/ldmx-sw #1109: Hcal SP Overlap

tomeichlersmith, 2022-10-31:
**Describe the bug**
@omar-moreno discovered this bug when updating the recoil tracker scoring planes.

It looks like the scoring plane around the back hcal is a little too close to the beam axis and extends into the side hcal volumes.

@EinarElen @cmantill 

https://github.com/LDMX-Software/ldmx-sw/pull/1100#issuecomment-1295677668

[image]

EinarElen, 2022-11-08:
From what I can tell the main issue here for the back hcal isn't actually that the back hcal scoring plane is wrong but that some of the back hcal layers are misaligned. I think we missed this because we were looking at wireframes where this was hard to see, in a solid visualization like below it is more clear. 
[image]

Regarding the Side Hcal part, I'm less sure about how to approach it since I'm not too familiar with how the scoring planes are used. Either we could surround the back hcal in one set of planes and the side hcal with another or we could surround the entire hcal with one big box.

EinarElen, 2022-11-08:
@cmantill the offset here (appearing in both X/Y-oriented scintillator layers) seems to be `+ hcal_scintWidth/2.0`.

EinarElen, 2022-11-08:
For updating the scoring planes, I _think_ the most straight-forward approach would be to replace references to `back_hcal_dfoo` with `hcal_envelope_dfoo` in the scoring planes file.

cmantill, 2022-11-08:
I guess that it makes sense to surround the back hcal in one set of scoring planes. For the side hcal we can do a separate box but I guess that in that case we need to be careful with the Ecal scoring planes?

cmantill, 2022-11-08:
> hcal_scintWidth

Hi @EinarElen , ok great. I think that that should be because we are placing the plate of absorber directly but then the scintillator bars need to be centered at `hcalScintWidth/2.0`. Can you add that to https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Detectors/data/ldmx-det-v14/hcal.gdml#L274 and https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Detectors/data/ldmx-det-v14/hcal.gdml#L285 ?

EinarElen, 2022-11-08:
Right, I'll make a PR with both of those changes

EinarElen, 2022-11-08:
Should the change be at those places or in the `back_hcal_startX_scint`/`back_hcal_startY_scint`-variables?

PR #1119 (2022-11-08): Iss1109
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
_Hint_: Use the phrase '_This resolves #< issue number >_' so that they are linked automatically.
This resolves #1109 
## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.
Ran visualization with the full detector including scoring planes without getting any overlaps. 
[image]
Below is a zoomed-in version on the back hcal showing that the layers aren't mismatched anymore 
[image]
closed 2022-11-08 by tomeichlersmith