LDMX-Software/ldmx-sw #1117: Hcal geometry condition updates + refactoring

EinarElen, 2022-11-08:
Given the new and more complicated hcal geometry, there are more things we want to be able to ask the geometry condition. I want to take the opportunity when adding these to also try and clean things up in there a bit. @ehrlich-uva feel free to suggest other things you might need below

- [ ] Scintillator bar length from HcalID 
- [ ] Scintillator bar orientation from HcalID (no longer only relevant for the back hcal...)

EinarElen, 2022-11-08:
For the scintillator bar orientation, I'm thinking maybe we can get away with not handling that yet. Currently, only the v14 geometry has a side-hcal which is relevant for Ralf's scintillation simulation so... We can survive with handling it there

ehrlich-uva, 2022-11-08:
Is there a function that returns the version number of the geometry? If so, we can use some if/thens to use the older HcalDigitization methods for older geometry versions.

EinarElen, 2022-11-08:
There isn't currently, but i still think it would be valuable to have two distinct digiproducers and in that case we maybe wouldn't need it

ehrlich-uva, 2022-11-08:
Yes, we can keep both digiproducers.

EinarElen, 2023-04-27:
I noticed now that I had forgotten to add the original feature that this issue was about :facepalm:. I added all the parts to the python config but no way of accessing them from the HcalGeometry objects. I'll add and test them ASAP, but this shouldn't make a difference w.r.t. the validation tests.

PR #1135 (2023-05-10): Iss1117
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1117. This primarily means providing the length of a scintillator bar given an HcalID from the geometry condition. Since the side hcal for v14 has different length of bars depending on the layer number, this needs to be a function of both section and layer.

Example use would be 
```C++ 
auto barLength {hcalGeometry.getScintillatorLength(id); 
```

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that t [... cut]
closed 2023-05-10 by EinarElen