LDMX-Software/ldmx-sw #1123: ECal v14 Validation Discussion

tomeichlersmith, 2022-11-11:
Just isolating the ECal discussion here because it may be too much for other folks...

@vdutta thoughts?

After the reco updates in #1115 , ECal is looking pretty good. The legend title refers the commit of trunk I'm running on (except for v3.0.0, that is v3.0.0). These plots were generated from 10k event samples of 4GeV electrons shot directly into the center of the ECal, skipping all upstream tracker/trigscint.

## Observations
- With the updated layer weights (and silicon thickness), the total recon energy is replicated well.
- Both average layer hit and deepest layer hit shift forward. I have no physical explanation for this.
- The number of readout hits decreases. I don't know why this is, but it could be due to the change in how the noise RMS is being applied, I should double check that it is being applied correctly.
- The total isolated energy shifts up again. I believe this is due to the fact that the support box _does not_ have an aluminum plate in front of the first sensitive layer (like in previous geometries) so there are a lot more events where the beam electron goes through only one cell in the first layer.
- The layer, x position, y position, and transverse position standard deviation all broaden slightly (here transverse position standard deviation is called shower RMS). I think this _could_ be due to the introduction of more material and the increase in the separation between sensitive layers.
- The max single-cell energy dep broadens slightly.

## Plots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

vdutta, 2022-11-11:
Thanks for the plots @tomeichlersmith! A couple of questions:

- Is the shift in the total isolated energy between v3.0.0 v12 and current v12 due to the fix you made in the calculation?
- The reduction in number of readout hits is indeed a bit confusing to me, so checking how the noise RMS is being applied would indeed be good. Is it the same between v12 and v14 in v3.2.0-gamma? If so, then the shift in the N readout hits would appear to be coming from the changes in the geometry and material, but it's quite significant. The shift in the avg layer hit and deepest layer hit distributions could be related to whatever's causing that.

tomeichlersmith, 2022-11-11:
- Yes, that is due to the calculation fix in #1079 
- The noise RMS should be the same between v12 and v14 in v3.2.0-gamma, but that is another thing to double check.

tomeichlersmith, 2022-11-15:
@vdutta The shift in the number of hits originates at the simulation-level and there is a suspiciously big difference between v12 right now and v12 from v3.0.0. Going to look into that...

[image]

tomeichlersmith, 2022-12-01:
## Left To Do
- [ ] Check that extra late-time sim hits are just extra soft particles (aligns with hypothesis that the extra air+absorber is affecting things)
- [ ] Include ldmx-sw-v3.0.0, geo-v12 in plots to make sure v3.2.0-geo-v12 doesn't have any surprises
- [ ] Tune second order energy correction parameter for v14 using 1M plain electrons

tomeichlersmith, 2022-12-02:
Even without the the tuning of the second order energy correction parameter, things are looking really good. I have developed a set of slides that are uploaded here as future reference. They contain all the plots I'm looking at as well as my comments about them.

[ecal_v14_geo_valid.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10144626/ecal_v14_geo_valid.pdf)

After my 1M plain electron sample generation completes, I can calculate the second order energy correction, put that into the config and ECal will be ready for v3.2.0.

tomeichlersmith, 2022-12-05:
Doing a normal distribution where the low and high sides have different deviations seems to give a better fit. If I wanted to spend some more time on the stats, I could look at how to avoid the discontinuity at the mean, but I think this looks good for this purpose.

[image]

# Code
```python
import uproot
import numpy as np
from scipy.stats import norm
import scipy
%matplotlib inline
import matplotlib.pyplot as plt
import mplhep
plt.style.use(mplhep.style.ROOT)

t = uproot.open('data/v3.2.0-gamma-14-geb86ea1/plain/type_events_sim_plain_geometry_v14_events_1000000_run_1.root:LDMX_Events')
tre = t['EcalVeto_valid/summedDet_'].array(library='np')

vals, edges, p = plt.hist(tre,bins='auto', range=(0,8000), density=True,
                         histtype='step',linewidth=2, label='1M Plain 4GeV')
centers = (edges[1:]+edges[:-1])/2

def two_sided_normal(x, mu, low_sig, high_sig) :
    low_side = norm.pdf(x[x<mu], mu, low_sig)
    high_side = norm.pdf(x[x>=mu], mu, high_sig)
    return np.concatenate((low_side,high_side))

# filter to reasonable values, specifically dropping high energy events
# from wacky hits late in the detector and low energy events probably
# due to PN interactions
high_cut = 6000
low_cut = 2500
tre_f = tre[(tre < high_cut)&(tre > low_cut)]

mean = tre_f.mean()
stdd = np.sqrt(((tre_f - mean)**2).mean())

bin_selection = (centers < high_cut) & (centers > low_cut)

optim, cov = scipy.optimize.curve_fit(two_sided_normal, centers[bin_selection], vals[bin_selection], 
                                      sigma=np.sqrt(vals[bin_selection]), p0=[mean,stdd,stdd])

plt.plot(centers, norm.pdf(centers, mean, stdd), 
         label=f'$\mu = {mean:.1f}$MeV $\sigma = {stdd:.1f}$MeV')
plt.plot(centers, two_sided_normal(centers, *optim), 
         label=f'$\mu = {optim[0]:.1f}$MeV\n$\sigma_l = {optim[1]:.1f}$MeV\n$\sigma_h = {optim[2]:.1f}$MeV')
plt.yscale('log')
plt.ylim(1e-9,1e-2)
plt.legend(loc='upper left')
plt.xlabel('Total Rec Energy [MeV]')
plt.ylabel('Event Fraction')
plt.show()
```

closed 2022-12-06 by tomeichlersmith