LDMX-Software/ldmx-sw #1137: Allow more than one type of G4 logical volume in the TS SDs

bryngemark, 2022-12-06:
**Is your feature request related to a problem? Please describe.**
When including vertical bars in some TS geometries, with the new way of attaching sensitive detector volumes (not in the gdml, but in SimCore code), I can't have all hits in a trigger pad module belong to the same collection anymore, since not all bars belong to the same logical volume. 

**Describe the solution you'd like**
Allow more than one type of logical volume to be added to the TS SDs, such that there is one SD per pad/module. This is done successfully for example in the HCal implementation. 

**Describe alternatives you've considered**
We could also use one horizontal, one vertical hint collection, rather than distinguishing them by channel ID as is done in my original vertical bar implementation (that I never pushed). This would mean running digi and clustering twice per pad, and rewriting tracking. I am not all that fond of this idea as it changes all workflows just because you switch between two geometries, and adds no real value in itself.

tomeichlersmith, 2024-05-13:
Is this resolved?

The code as written makes me think that the same SD is being attached to multiple logical volumes (ones that contain the `vol_name_` and don't have the scoring plane prefix `sp_`).

https://github.com/LDMX-Software/ldmx-sw/blob/3ca322e8b3b98b07e8478fb543b3f39d11615105/SimCore/include/SimCore/SDs/TrigScintSD.h#L27-L40

bryngemark, 2024-05-14:
it has not been solved but thanks for reminding me of this issue! pulling in Elizabeth and @rodwyer100 here for reference

tvami, 2024-09-21:
@rodwyer100 can I assign you to this issue?
@bryngemark I'm sorry for my ignorance on the TS, but is the TS geometry with the vertical bars in v14, or is this issue blocking that inclusion?