LDMX-Software/ldmx-sw #1138: v3.2.0 validation sample feedback

bryngemark, 2022-12-19:
This issue is for keeping track of results checking the v3.2.0 release validation samples. 

Once your system is happy/have submitted related issues and merged their fixes, check the system box below.

Systems ok:
- [ ] Trigger Scintillator
- [ ] Tracker
- [ ] Ecal 
- [ ] Hcal
- [ ] Trigger

Feel free to upload plots and comments in this thread, or link to subsystem issues as you see fit. Also add/remove assignees as relevant to your system. Please add requests for specific additional samples needed for your systems to this thread. 

@pbutti @omar-moreno, I add the tracker here in case you want to run any tracking tests with the validation samples. If you'd rather pass this time, just check your box.

bryngemark, 2023-01-02:
[TS validation plots 1e](https://github.com/LDMX-Software/ldmx-sw/files/10332274/TSplots_v12v14.pdf)
[TS validation plots 2e](https://github.com/LDMX-Software/ldmx-sw/files/10332273/TSplots_v12v14_2e.pdf)

This all looks reasonable given the geometry changes. Good enough to go.

tomeichlersmith, 2023-01-06:
[signal-1-overlay-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362318/signal-1-overlay-ecal-shower-feats.pdf)
[signal-1-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362319/signal-1-ecal-shower-feats.pdf)
[signal-0.001-overlay-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362320/signal-0.001-overlay-ecal-shower-feats.pdf)
[signal-0.001-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362321/signal-0.001-ecal-shower-feats.pdf)
[ecal-pn-overlay-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362322/ecal-pn-overlay-ecal-shower-feats.pdf)
[2e-inclusive-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362323/2e-inclusive-ecal-shower-feats.pdf)
[1e-inclusive-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362324/1e-inclusive-ecal-shower-feats.pdf)
[1e-ecal-pn-ecal-shower-feats.pdf](https://github.com/LDMX-Software/ldmx-sw/files/10362325/1e-ecal-pn-ecal-shower-feats.pdf)

To me this looks reasonable. There is some funkiness in the signal distributions, but I think we are good to go because signal is generally faster to create compared to the much larger bkgd samples we need to generate (and I'm not sure if this funkiness is due to my own mishandling of the files that make up the sample).

I will look more into the signal samples and see if it is a bug with ldmx-sw. If it is a bug, I will create an issue for that and we can have a patch release for signal simulation.

tomeichlersmith, 2023-01-09:
Looking at just one of the signal files shows that my hypothesis was correct. We observe the dual peaks and the peak at the beam energy is from the events where there is no A' sim particle.

[image]

## Jupyter Notebook Code

```python
%matplotlib inline
import matplotlib.pyplot as plt
import mplhep
plt.style.use(mplhep.style.ROOT)

import uproot

with uproot.open('mc_v14-4.0GeV-1e-signal_W_noDecay_sim_run10000000_t1671225529.root') as f :
    # read in particle PDGs and total det energy
    df = f['LDMX_Events'] \
        .arrays(['SimParticles_signal.second.pdgID_','EcalVeto_signal/summedDet_'],
                library='pd') \
        .rename(columns={'SimParticles_signal.second.pdgID_':'sim_pdgs',
                         'EcalVeto_signal/summedDet_':'tot_rec_energy'})

# reduce the list of all PDGs down to just a count of the number of A' in each "entry" (event)
# just call mean on the tot_rec_energy which has the same value for all subentries in a single event
df = df.agg({'sim_pdgs' : lambda series : series.eq(622).groupby('entry').sum(),
        'tot_rec_energy' : lambda series : series.groupby('entry').mean()
       })

# plot
for name, sl in [('DB Occurred',df['sim_pdgs']==1),('No DB',df['sim_pdgs']!=1)] :
    plt.hist(df[sl]['tot_rec_energy'], label=name,
             bins = 50, range = (0,5000), histtype='step', linewidth = 2)
plt.yscale('log')
plt.ylabel('Event Count')
plt.xlabel('Total Rec Energy [MeV]')
plt.legend()
plt.show()
```

bryngemark, 2023-01-09:
Cool! What to do, increase the number of attempts?

tomeichlersmith, 2023-01-09:
I'm not sure, I'm confused on why this doesn't happen for v12 so I want to check if there is a bug in the sim filters. If no bugs, then yes, increasing the tries should lessen this issue (but won't eliminate it).

tomeichlersmith, 2023-01-10:
I've seen some funky behavior and I don't think it is a number-of-attempts problem anymore so I'm promoting this to its own issue https://github.com/LDMX-Software/ldmx-sw/issues/1139

tomeichlersmith, 2023-09-13:
Closing this while doing triage for the v3.3.0 release. Feel free to reopen if further discussion on these samples is needed.

closed 2023-09-13 by tomeichlersmith