LDMX-Software/ldmx-sw #1139: Signal Filter Fall-Through

tomeichlersmith, 2023-01-10:
**Describe the bug**
Some events _without_ a successfully simulated dark photon make it through our simulation filters. This defeats the purpose of the filters and points to an edge case that we did not think of when designing the filters.

**To Reproduce**
With v3.2.0, run the following config
```python
from LDMX.Framework import ldmxcfg
p = ldmxcfg.Process('target_db')
from LDMX.Biasing import target, util
from LDMX.SimCore import makePath
import LDMX.Ecal.EcalGeometry
import LDMX.Hcal.HcalGeometry
import LDMX.Ecal.digi as ecal_digi
import LDMX.Ecal.vetos as ecal_veto
import LDMX.Ecal.ecal_hardcoded_conditions
s = target.dark_brem(
        10., #MeV - mass of A'
        makePath.makeLHEPath(10.), #str - full path to directory containing LHE vertex files
        'ldmx-det-v14' , #name of geometry to use
        )
p.sequence = [s,
    ecal_digi.EcalDigiProducer(),
    ecal_digi.EcalRecProducer(),
    ecal_veto.EcalVetoProcessor()
    ]
p.maxEvents = 1000
p.maxTriesPerEvent = 100
p.outputFiles = [ 'target_db.root' ]
```
Then we can plot the 1k events and see two events without a dark photon.
```python
with uproot.open('target_db.root') as f :
    # read in particle PDGs and total det energy
    df = f['LDMX_Events'] \
        .arrays(['SimParticles_target_db.second.pdgID_','EcalVeto_target_db/summedDet_'],
                library='pd') \
        .rename(columns={'SimParticles_target_db.second.pdgID_':'sim_pdgs',
                         'EcalVeto_target_db/summedDet_':'tot_rec_energy'})

# reduce the list of all PDGs down to just a count of the number of A' in each "entry" (event)
# just call mean on the tot_rec_energy which has the same value for all subentries in a single event
df = df.agg({'sim_pdgs' : lambda series : series.eq(622).groupby('entry').sum(),
        'tot_rec_energy' : lambda series : series.groupby('entry').mean()
       })

# plot
for name, sl in [('DB Occurred',df['sim_pdgs']==1),('No DB',df['sim_pdgs']!=1)] :
    plt.hist(df[sl]['tot_rec_energy'], label=name,
             bins = 50, range = (0,5000), histtype='step', linewidth = 2)
plt.yscale('log')
plt.ylabel('Event Count')
plt.xlabel('Total Rec Energy [MeV]')
plt.legend()
plt.show()
```
[image]

The `df` has two entries (index 893 and 593) that have no dark photons. The dataframe index starts at 0 while the event numbering starts at 1, so we are looking at events 894 and 594 in more detail.

I uncommented some debug printouts in the TaggerVetoFilter and TargetDarkBremFilter so I could make sure each event is ending with (a) a found A' or (b) a known reason to abort.

Both events end without reaching the maximum number of tries (100) or finding an A'.
### 594
```
[ TargetDarkBremFilter ]: (593) A' originated in target
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TaggerVetoFilter ]: (594) Primary energy fell below threshold. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (594) Primary electron did not create A'. Aborting event.
[ TaggerVetoFilter ]: (594) Primary energy fell below threshold. Aborting event.
[ TargetDarkBremFilter ]: (595) Primary electron did not create A'. Aborting event.
```
### 894
```
[ TargetDarkBremFilter ]: (995) Primary electron did not create A'. Aborting event.                                                                                                                                                                                             
[ TargetDarkBremFilter ]: (894) Primary electron did not create A'. Aborting event.                                                                                                                                                                                             
[ TargetDarkBremFilter ]: (894) Primary electron did not create A'. Aborting event.
[ TargetDarkBremFilter ]: (894) Primary electron did not create A'. Aborting event.
[ TaggerVetoFilter ]: (894) Primary energy fell below threshold. Aborting event.
[ TargetDarkBremFilter ]: (895) Found an A' weighted 1.11592e-06 originating in target with energy 3927.85 MeV.
[ TargetDarkBremFilter ]: (895) A' originated in target
```

Even weirder, `894` is _after_ event `995` ??? I'm also confused on why that is.

## Plan
I am going to debug this on branch `SimCore-83-external-g4db` because those developments are also related to the signal simulation and are pretty much done. 

https://github.com/LDMX-Software/SimCore/pull/84

tomeichlersmith, 2023-01-11:
@EinarElen pointed out that the Kaon's fall through may also be caused by a similar issue. https://github.com/LDMX-Software/ldmx-sw/issues/1069

tomeichlersmith, 2023-01-12:
I've re-run the test above on my updated branch and am seeing more events without a dark brem in it. A change was expected since the new implementation of the dark brem process includes an update to the cross section calculation.

All of the events get through to being stored by the `Simulator` but do not find an A' within the filter. It looks like the primary electron is doing some sort of hard interaction upstream of the target since the events without a DB also have end points where the primary electron does not end within the target.
[image]

tomeichlersmith, 2023-01-12:
I tried a few different re-designs of the TaggerVetoFilter but I couldn't get an improvement. What _does_ work but is not very fancy is to update the TargetDarkBremFilter to check for an A' at the end of the event. This manually catches those few events that sneak past the filtering earlier in the simulation.

The next logical question is
### Why doesn't this happen for v12?
I have two hypothesis that aren't mutually exclusive.
1. v12 has less material upstream of the target, so the rate of events where the primary electron does not reach the target (or misses the target to a side) is lower.
2. v12 did not have an air gap between the tagger and target regions. This air gap may be allowing some scattered primary electrons to "sneak" past the filter without going through the target.

## Conclusion
This means any simulation that uses the `TaggerVetoFilter` needs to have a secondary check that the primary particle is entering the target region at some point during the event. Simpler to implement but potentially costly at runtime is to have an end-of-event double check that the in-target process that we wanted to occur did in-fact occur.

tomeichlersmith, 2023-01-12:
Same plot as above but with updated biasing filter...

[image]

bryngemark, 2023-01-12:
thanks @tomeichlersmith. so are you saying you think the electron never hits the target, by scattering or otherwise? it would be great to know how often this happens, for v12 and v14. (potentially one could use scoring planes to check this?) also, what is the direction of the momentum at those negative endZ points?

the event number shuffling is potentially worth tracking more, at least checking if this is a one-off or recurring. (and if it changes between geometries...) i suppose just crawling the printouts for non-monotonous behavior is a start.

tomeichlersmith, 2023-01-13:
First the easy bit, I have not been able to reproduce the event-number-shuffling so I'm pretty confident it was a one-off occurrence (probably messiness from laziness on my part).

The direction of momentum at the negative endZ points is negative so they were traveling away from the target. I'm not sure how to quantify how often this happens since we are using many tries per event. Since we abort all of the "easy"/"simple" events, we naturally are seeing more complicated events. I mean, ignoring biasing, this effect is roughly 7 events out of 1000 (events) *1000 (tries per event) = 1M events.

I guess, to be transparent, I think understanding these 7 events would require some time with the large inclusive sample. Just because the primary electron bounces back doesn't mean it didn't transfer most of its energy to another electron which we would subjectively call the primary.

PR #1141 (2023-01-24): Update Dark Brem Process and Biasing Filter
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1139 by updating the biasing filter selecting signal events in the target.

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.
  - see #1139 and the sub-module PR for verification plots and comments
- [x] I attached any sub-module related changes to this PR.

### Related Sub-Module PRs
- https://github.com/LDMX-Software/SimCore/pull/84

We are  [... cut]
closed 2023-01-24 by tomeichlersmith