LDMX-Software/ldmx-sw #1146: TaggerVetoFilter inefficiency

tomeichlersmith, 2023-01-25:
## Description
This was initially observed when studying the signal generation using this filter. What I observed through some extra debug printouts (and using the `StepPrinter` user action) was that the _primary_ electron was not entering the target region. This did not "activate" the `TargetDarkBremFilter` which means it did not know to veto the event. The primary electron was found with $p_z < 0$ and $z < 0$ at its end point, so I think something caused the primary electron to back-scatter. We still see a near-beam-energy total reconstructed energy in the ECal however, so particles are still reaching it.

### Initial Hypothesis
Various small scatters bend the primary electron into the magnet which causes a much harder scatter (perhaps electron-nuclear or perhaps just a hard brem). The non-primary products of this hard scatter continue downstream into the ECal while the primary electron recoils back, avoiding the target and thus the `TargetDarkBremFilter`.

### Additional context
https://github.com/LDMX-Software/ldmx-sw/issues/1138#issuecomment-1376305770
https://github.com/LDMX-Software/ldmx-sw/issues/1139

## Plan
- Isolate and confirm issue: develop config where we can observe the TaggerVetoFilter failling to veto an event where the primary electron misses the target.
- Develop update: try modifications to TaggerVetoFilter to "catch" these types of events.

EinarElen, 2023-03-05:
While working on https://github.com/LDMX-Software/SimCore/issues/86, I ran into https://github.com/LDMX-Software/ldmx-sw/issues/1069 quite a lot which really made validating developments hard. Since I was already messing with the photonNuclear process, I decided to try _removing the PN process entirely_ to see what would happen. To my surprise, after 3204 aborted attempts the event passes the filters. I'm presuming this is related to this issue. 

Gamma physics change to try this: 
```c++
void GammaPhysics::ConstructProcess() {
  aParticleIterator->reset();

  // Loop through all of the particles and find the "gamma".
  while ((*aParticleIterator)()) {
    G4ParticleDefinition *particle = aParticleIterator->value();
    G4ProcessManager *pmanager = particle->GetProcessManager();
    G4String particleName = particle->GetParticleName();

    if (particleName == "gamma") {
      // Get the process list associated with the gamma.
      G4ProcessVector *vProcess = pmanager->GetProcessList();

      // Search the list for the process "photoNuclear".  When found,
      // change the calling order so photonNuclear is called before
      // EM processes. The biasing operator needs the photonNuclear
      // process to be called first because the cross-section is
      // needed to bias down other process.
      for (int iProcess = 0; iProcess < vProcess->size(); ++iProcess) {
        G4String processName = (*vProcess)[iProcess]->GetProcessName();
        if (processName == "photonNuclear") {
          pmanager->RemoveProcess(iProcess);
          // pmanager->SetProcessOrderingToFirst((*vProcess)[iProcess],
          //                                     G4ProcessVectorDoItIndex::idxAll);
        }
      }

      // Add the gamma -> mumu to the physics list.
      pmanager->AddDiscreteProcess(&gammaConvProcess);
    }
  }
}
```

With the config below: 
```python 
#!/bin/python
from LDMX.Framework import ldmxcfg
det = 14
run = 1
maxEvents = 5
thisPassName = "sim"
detector=f'ldmx-det-v{det}'
p = ldmxcfg.Process(thisPassName)

p.maxEvents = maxEvents
p.maxTriesPerEvent = 100000000000

#import LDMX.SimCore.photonuclear_models as pns
from LDMX.Biasing import ecal
from LDMX.DQM import dqm
from LDMX.SimCore import generators as gen
import LDMX.Ecal.EcalGeometry
import LDMX.Ecal.ecal_hardcoded_conditions
import LDMX.Hcal.HcalGeometry
import LDMX.Hcal.hcal_hardcoded_conditions
from LDMX.SimCore import simulator
from LDMX.Biasing import filters
from LDMX.Biasing import util
from LDMX.Biasing import include as includeBiasing

sim = simulator.simulator('ie')
sim.setDetector(f'ldmx-det-v{det}')

sim.generators = [gen.single_4gev_e_upstream_tagger()]
sim.beamSpotSmear = [20., 80., 0.] #mm
[... 87 lines cut ...]
 Step 59 { Energy: 8.26831 Track ID: 1 Particle currently in: World_PV Region: DefaultRegionForTheWorld Next volume: ldmx_magnet_PV Weight: 1 Children: }
 Step 60 { Energy: 8.26831 Track ID: 1 Particle currently in: ldmx_magnet_PV Region: MagnetRegion Next volume: World_PV Weight: 1 Children: }
 Step 61 { Energy: 8.26831 Track ID: 1 Particle currently in: World_PV Region: DefaultRegionForTheWorld Next volume: trig_scint_tagger_PV Weight: 1 Children: }
 Step 62 { Energy: 8.2683 Track ID: 1 Particle currently in: trig_scint_tagger_PV Region: trig_scint Next volume: World_PV Weight: 1 Children: }
 Step 63 { Energy: 8.2683 Track ID: 1 Particle currently in: World_PV Region: DefaultRegionForTheWorld Next volume: ldmx_magnet_PV Weight: 1 Children: }
 Step 64 { Energy: 8.2683 Track ID: 1 Particle currently in: ldmx_magnet_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 65 { Energy: 8.0327 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: 22 }
 Step 66 { Energy: 7.84064 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: 22 }
 Step 67 { Energy: 7.35024 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: 22 }
 Step 68 { Energy: 6.04606 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 69 { Energy: 5.69689 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: 22 }
 Step 70 { Energy: 5.03528 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 71 { Energy: 4.4176 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 72 { Energy: 3.69235 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 73 { Energy: 3.09606 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 74 { Energy: 2.47338 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 75 { Energy: 1.80446 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
 Step 76 { Energy: 1.14288 Track ID: 1 Particle currently in: shim_right_PV Region: MagnetRegion Next volume: shim_right_PV Weight: 1 Children: }
[... cut ...]
As a _really_ dumb way to test that this is the issue, I went ahead and added a global (through the run manager) that was set to false at the start of each event which was set to true if we reached the relevant part of the filter and then just aborted the event in the simulator if it wasn't touched. Obviously, this isn't a solution but it does patch both this and the next events that would have failed for the same reason (at the fourth one, the step printer crashes but... that's a different issue).

EinarElen, 2023-03-05:
A simple, albeit not very elegant, solution to this is to make the TaggerVetoFilter an EVENT action as well as a STEPPING action, add a field to the event info which corresponds to whether or not we ever entered the tagger region with the primary that we set in the stepping action and then in the EndOfEventAction just checking this and aborting if it fails. I ran 1000000 such events without a PN process and it worked without issue.

EinarElen, 2023-03-05:
Interestingly, this still isn't enough to fix https://github.com/LDMX-Software/ldmx-sw/issues/1069... However, that issue seems to be unrelated, it can fail _even when the PN products are present in the output_ :(

closed 2023-03-06 by tomeichlersmith