LDMX-Software/ldmx-sw #1149: TS Scoring Planes are not being attached

tomeichlersmith, 2023-02-07:
**Describe the bug**
Running on current `trunk` (which is v3.2.1 + updated gold histograms) with high verbosity, we see in the basic simulation that there is no line for `Attaching trigscint_sp` to any of the TS scoring plane logical volumes.

**To Reproduce**
Steps to reproduce the behavior:
1. Build trunk and install it
2. Run basic: `ldmx fire ldmx-sw/SimCore/test/basic.py`
3. See the `sp_trigScint_PV` in the `ParallelWorld` printouts but no trig scint scoring plane SDs in the `DetectorConstruction` printouts.

**Desired behavior**
The scoring plane SDs are attached to the trig scint logical volumes as expected.

**Additional Context**
Expanded from discussion starting with https://github.com/LDMX-Software/ldmx-sw/pull/1121#issuecomment-1408734768

Possible fix within SimCore: https://github.com/LDMX-Software/SimCore/pull/87

tomeichlersmith, 2023-02-07:
@bryngemark can you double check that my patch in SimCore works?

```
cd ldmx-sw
git fetch
git checkout trunk
git pull
cd SimCore
git fetch
git checkout patch-1104
cd ..
ldmx clean src
git submodule update --recursive --init
ldmx cmake -B build -S .
cd build
ldmx make install
ldmx fire ../SimCore/test/basic.py
```

bryngemark, 2023-02-14:
ok i finally had a minute (an hour) for this.

running the indicated config i see:
```
[ ParallelWorld ]: Adding : sp_trigScint_PV to parallel world.
[ ParallelWorld ]: Adding : sp_trigScint_PV to parallel world.
[ ParallelWorld ]: Adding : sp_trigScint_PV to parallel world.
[ ParallelWorld ]: Adding : sp_trigScint_PV to parallel world.
[ ParallelWorld ]: Adding : sp_trigScint_PV to parallel world.
[ ParallelWorld ]: Adding : sp_trigScint_PV to parallel world.
```

but the collection `TrigscintScoringPlaneHits_test` is (still) empty. All the other collections (including TS simhits and other scoring plane collections) are non-empty.

bryngemark, 2023-02-14:
from the above it doesn't look like the capitalization is working. i.e. the Scint is still capitalized in the collection name but not in the sp name.

bryngemark, 2023-02-14:
oh nevermind. the order of submodule operations above obviously undoes the checkout of the patch. checked it out again and rebuilt. now the collection is there :)

PR #1150 (2023-02-14): bump Framework to v1.3.2
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This integrates a patch written by @EinarElen for Framework and resolves #1149  by patching some too-quick-coding on my part.

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful.
- [x] I attached any sub-module related changes to this PR.

## Submodule PR
- https://github.com/LDMX-Software/Framework/pull/63
- https://github.com/LDMX-Software/SimCore/pull/87
closed 2023-02-14 by tomeichlersmith