LDMX-Software/ldmx-sw #1155: An 8 GeV version of v14 detector geometry

bryngemark, 2023-04-21:
It would be great to get going on some 8 GeV samples that we can compare apples-to-apples to the latest and greatest 4 GeV samples.

**Describe the solution you'd like**
The tagger tracker and TS upstream pads need to move to be aligned with the 8 GeV beam which has different rigidity.

**Describe alternatives you've considered**
ehh

**Additional context**
If there is a simple calculation to fix this, whoever has it, please add it to the issue :)

tomeichlersmith, 2023-08-10:
This is also something I've wanted to have for an 8GeV EaT analysis! My general thoughts are to expand our current naming scheme to include the expected beam energy.

The detector-name would then be `ldmx-<detname>-v<version_number>-<beam_energy>` for example
`ldmx-det-v14-8GeV` would be this new detector. Any detector _without_ a `-<beam_energy>` suffix would be assumed to be for 4GeV _or_ whose design does not depend on the beam energy (like the test beam `hcal-prototype` geometry). This would then require re-naming some of the other detectors which already have a suffix.
- `ldmx-det-v9-ti-target` -> `ldmx-ti-target-v9`
- `ldmx-det-v12-lyso` -> `ldmx-lyso-v12`

I can also write up a README for the Detectors/data directory describing this scheme so future users know what we are doing and how we are trying to stay organized.

EinarElen, 2023-08-10:
Sounds ressonable to my mind. If we are working on making changes to the detector geometries, one thing that has been on my wishlist for a while and which tends to trap newcommers would be to make the build system better at recognizing changes to the geometry. 

Currently, if I make changes to a gdml file and do make install, most of the time cmake won't notice that anything has changed so the installed gdml files won't change. What beginners tend to do is to go in and try and make the changes in the install directory instead. This works, untill you need to recompile which will overwrite what you have in the install directory. 

I think what's happening is that cmake copies the gdml sources to the build directory when we configure and then it uses whatever it copied as the source when installing

Also, if we are changing the naming schemes we should make sure that all the detectors_valid stuff in the geometry configurations are still matching.

PR #1185 (2023-08-17): an 8 gev version of v14 detector geometry
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1155 primarily but it also implements other changes.
- write up and enforce a naming convention for our detectors
- patch the primary generators for the electron beam at different energies (SimCore update)
  - **Note**: I expect this change to cause the PR validation plots to fail since it moves the location of the beam and therefore will cause the hits in the detector to be modified slightly.

### To Do
- [x] update Detectors cmake code so it ca [... cut]