LDMX-Software/ldmx-sw #1163: Wrong collection by default in electron counter?

EinarElen, 2023-05-17:
**Describe the bug**
I was looking a bit at the output of the PR jobs and half of the output is just 
```sh
 [ ElectronCounter ] 4 : Attemping to use non-existing input collection TriggerPadTracks_ to count electrons! Exiting.
```

Seems to be 
https://github.com/LDMX-Software/ldmx-sw/blob/1d6239b49687bc59c1efa60ad2a1fec5c0900027/Recon/python/electronCounter.py#L19-L23 

In LDCS jobs, the collection is called 
https://github.com/LDMX-Software/LDCS/blob/50570eab9b24100fd9be66221adc2a720a1260ae/productionScripts/v3.2/ldmxjob-1e-3.2.0.py.tpl#L119-L122
```python
eCount.input_collection="TriggerPadTracksY"
```

I'm not super familiar with the details of the TS/electron counting components so it isn't obvious what to do to deal with it

bryngemark, 2023-05-17:
for posterity: the LDCS way is the way to go, the default should be updated to include the `Y` since, for a while now, we also have the capability to run tracking in _x_ (even if there is no official geometry where that is feasible yet).

EinarElen, 2023-05-17:
Should we make that update when updating gold histograms for 
https://github.com/LDMX-Software/ldmx-sw/pull/1160

bryngemark, 2023-05-29:
> Should we make that update when updating gold histograms for
> #1160

Yes I think so

bryngemark, 2023-06-10:
@EinarElen is now a good time?

closed 2023-08-18 by tomeichlersmith