LDMX-Software/ldmx-sw #1172: Bug and typo in HcalDQM

EinarElen, 2023-06-18:
**Describe the bug**
This was caught thanks to the improved GCC warnings with the new container :) 

In https://github.com/LDMX-Software/ldmx-sw/blob/ad86bf811769fa94411712f7ef82ba6e2efdea00/DQM/src/DQM/HcalInefficiencyDQM.cxx#L30-L37 

only the back check actually tests against the failure condition, the others are doing an implicit conversion bewteen bool and int. 

Also, there as a minor typo in the python config 
https://github.com/LDMX-Software/ldmx-sw/blob/ad86bf811769fa94411712f7ef82ba6e2efdea00/DQM/python/dqm.py#L132-L137
should of course have been an f-string and not just a string that starts with f :)

EinarElen, 2023-06-18:
There are a couple of similar warnings from other places
```sh
/home/einarelen/ldmx/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx: In member function 'virtual void trigscint::QIEDecoder::produce(framework::Event&)':
/home/einarelen/ldmx/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:150:58: warning: narrowing conversion of '(int)(((unsigned char)((int)(flags >> 3))) & ((unsigned char)((int)mask8<1>::m)))' from 'int' to 'bool' [-Wnarrowing]
  150 |   bool isCIDskipped { (flags >> QIEStream::CID_SKIP_POS) & mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/einarelen/ldmx/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:151:59: warning: narrowing conversion of '(int)(((unsigned char)((int)(flags >> 2))) & ((unsigned char)((int)mask8<1>::m)))' from 'int' to 'bool' [-Wnarrowing]
  151 |   bool isCIDunsync { (flags >> QIEStream::CID_UNSYNC_POS) & mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/einarelen/ldmx/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:152:60: warning: narrowing conversion of '(int)(((unsigned char)((int)(flags >> 1))) & ((unsigned char)((int)mask8<1>::m)))' from 'int' to 'bool' [-Wnarrowing]
  152 |   bool isCRC1malformed{ (flags >> QIEStream::CRC1_ERR_POS) & mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/einarelen/ldmx/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:153:60: warning: narrowing conversion of '((((int)flags) >> 0) & ((int)mask8<1>::m))' from 'int' to 'bool' 
-Wnarrowing]
  153 |   bool isCRC0malformed{ (flags >> QIEStream::CRC0_ERR_POS) & mask8<QIEStream::FLAG_SIZE_BITS>::m};
``` 
Its not immediately obvious to me whether this implicit conversion is ok or not, maybe @bryngemark knows?

bryngemark, 2023-06-19:
hmmm, maybe what we're after there is an `&&` ? it should become a bool in the end so that part is fine but we'd like it to be the output of the right logic.

closed 2023-10-31 by tomeichlersmith