LDMX-Software/ldmx-sw #1176: Static analysis meta-issue

EinarElen, 2023-06-28:
Similar to #1166 I'm creating this issue to track warnings from compilers or static analysis.  Generally these aren't urgent to patch (I'll note if/when I think they need more immediate attention) but for future validation/PR workflows reducing the amount of these will allow us to use warnings that currently generate too many false-ish positives (true positives but probably not all that impactful) to be useful. 

To do this I'm currently building a custom version of the container environment with some additional tools (clang, clang-tidy so-far). If this turns out to be something we want to use in the future we could consider adding to the regular container. I'm also adding some additional support for adding additional warnings in the cmake module while working on https://github.com/LDMX-Software/cmake/issues/17. 

Whatever we decide for an approach for #1168 probably should guide if/how we would apply this.

EinarElen, 2023-06-29:
One of the cases of using uninitialized memory comes from the `SimCalorimeterHit` class. Since several member variables don't have a default value and they are generally constructed like
```C++
HitType hit;
hit.setA(A);
hit.setB(B);
```
It is easy to miss out on parts of the hit. This would have been caught by the Hcal/Ecal digi pipeline tests for the noise member but both of them are doing `CHECK_FALSE` rather than `REQUIRE_FALSE` so the test doesn't actually fail...

EinarElen, 2024-09-05:
There are a couple of things you can run at compile time at the moment. I've checked off the ones I think are mostly clean but a lot of things have changed since them (including that you have patched a decent chunk of these on your own 😄). 

There are a couple of things we can do when it comes to static analysis, and ideally we would be running our CI with as many of these enabled as possible. I would probably try and squash them in steps but if you feel brave you can run them all at once, feel free. To make warnings a hard error, you can add `-DWARNINGS_AS_ERRORS=ON`.

```sh
// Build with clang and check that we don't get any warnings 
just configure  -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang
just build 
// Clean build directory and increase warning levels and add LTO, see CompilerWarnings.cmake/InterProceduralOptimization.cmake
rm -rf build
just configure -DADDITIONAL_WARNINGS=ON -DENABLE_LTO=ON  -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang
just build 
// Clean build directory and add clang-tidy (see StaticAnalysis.cmake)
rm -rf build
just configure -DADDITIONAL_WARNINGS=ON  -DENABLE_CLANG_TIDY=ON  -DENABLE_LTO=ON  -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang
just build
// And switch back to a build with GCC
rm -rf build 
just configure -DADDITIONAL_WARNINGS=ON  -DENABLE_CLANG_TIDY=ON  -DENABLE_LTO=ON  -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang
just build
```

I think that if we can get a clean build with these, we are in a really good place. I would start with clang because, in general, i think the builds with clang are a bit faster to run and a lot of the warnings will overlap between the two. There are some warnings that are from ROOT and ACTS and... I don't know what we can do about those

EinarElen, 2024-09-05:
Instead of using warnings as errors while working on this, I would pipe stderr/stdout to a logfile :)

tvami, 2024-09-05:
Thanks @EinarElen !

> There are some warnings that are from ROOT and ACTS and... I don't know what we can do about those

Yes I'm a bit worried about what we should do if we enable these settings at the CI (after the ldmx-sw problems are fixed)

EinarElen, 2024-09-05:
I would suggest having an equivalent to the gold logs for the compilation steps. So them being there isnt the end of the world, but if any new ones show up we'll be notified

tvami, 2024-09-06:
OK so running
```
just configure  -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang
just build 
```
 gives me this list
 - issue with TString.h, RStringView.hxx --> Outside of our scope
 - nested namespace --> I guess I can fix this, but it should be fine too as is
```
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Framework/include/Framework/Performance/Timer.h:10:20: warning: nested namespace definition is a C++17 extension; define each namespace separately [-Wc++17-extensions]
namespace framework::performance {
                   ^~~~~~~~~~~~~
                    { namespace performance
1 warning and 12 errors generated.
```
 - missing `optional` --> I think this is fine in newer versions of clang, and we are using an older version?
```
In file included from /sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/build/Framework/EventDict.cxx:68:
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Tracking/include/Tracking/Event/Track.h:86:8: error: no template named 'optional' in namespace 'std'
  std::optional<TrackState> getTrackState(TrackStateType tstype) const {
  ~~~~~^
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Tracking/include/Tracking/Event/Track.h:88:45: error: no member named 'optional' in namespace 'std'
      if (ts.ts_type == tstype) return std::optional<TrackState>(ts);
                                       ~~~~~^
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Tracking/include/Tracking/Event/Track.h:88:54: error: 'TrackState' does not refer to a value
      if (ts.ts_type == tstype) return std::optional<TrackState>(ts);
                                                     ^
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Tracking/include/Tracking/Event/Track.h:57:10: note: declared here
  struct TrackState {
         ^
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Tracking/include/Tracking/Event/Track.h:90:17: error: no member named 'nullopt' in namespace 'std'
    return std::nullopt;
           ~~~~~^
```

I think we should add 
```
set(CMAKE_CXX_STANDARD 17)
```
into `ldmx-sw/CMakeLists.txt` and run this command like that (then these above are gone and we have some other stuff coming in). Do you agree @EinarElen @tomeichlersmith ?

tvami, 2024-09-06:
With `set(CMAKE_CXX_STANDARD 17)` the list is the following.
[log-part1.txt](https://github.com/user-attachments/files/16912689/log-part1.txt)

tvami, 2024-09-06:
Another question to Tom
```
/sdf/home/t/tamasvami/iss1176-part1-uninitialized/ldmx-sw/Ecal/src/Ecal/EcalRawDecoder.cxx:133:48: warning: & has lower precedence than ==; == will be evaluated first [-Wparentheses]
      bool rid_ok = (w >> (shift_in_word + 7)) & packing::utility::mask<1> == 1;
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
was this meant to be an `&& (packing::utility::mask<1> == 1)` ?

tomeichlersmith, 2024-09-09:
For the first one, we are checking if a bit is set, so
```diff
-       bool rid_ok = (w >> (shift_in_word + 7)) & packing::utility::mask<1> == 1;
+      bool rid_ok = ((w >> (shift_in_word + 7)) & packing::utility::mask<1>) == 1;
```
For the others, I wanted to show how the bit shift was being calculated so the numbers being summed should have parentheses around them so that they are added before applied as the shift.
```diff
-      word |= (1 << 12 + 1 + 6 + 8);  
+      word |= (1 << (12 + 1 + 6 + 8));  
```
```diff
-      word |= (fpga_id & packing::utility::mask<8>) << 12 + 1 + 6;   // FPGA
+      word |= (fpga_id & packing::utility::mask<8>) << (12 + 1 + 6);   // FPGA
```

tvami, 2024-09-09:
Thanks Tom, I'll have this included in the next batch!

tvami, 2024-09-14:
@EinarElen I'm in the last bit of this, but your last line is the same as the one before. What did you mean to have there?

EinarElen, 2024-09-14:
Was supposed to be with the clang/clang++ parts removed!

tvami, 2024-09-14:
> Was supposed to be with the clang/clang++ parts removed!

ok I updated your msg to 
```
just configure -DADDITIONAL_WARNINGS=ON  -DENABLE_CLANG_TIDY=ON  -DENABLE_LTO=ON 
```

closed 2024-09-17 by tvami