LDMX-Software/ldmx-sw #1199: Narrowing conversion warnings by the compiler

tvami, 2023-09-14:
**Describe the bug**
Inspired by the comment in https://github.com/LDMX-Software/ldmx-sw/pull/1198#issuecomment-1717288314 when I compile ldmx-sw I see the following narrowing conversions in ldmx-sw:

In `HcalGeometry.cxx`
```
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx: In member function 'void ldmx::HcalGeometry::buildStripPositionMap()':
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:156:31: warning: narrowing conversion of 'section' from 'int' to 'unsigned int' [-Wnarrowing]
  156 |         const ldmx::HcalID id{section, layer, strip};
      |                               ^~~~~~~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:156:40: warning: narrowing conversion of 'layer' from 'int' to 'unsigned int' [-Wnarrowing]
  156 |         const ldmx::HcalID id{section, layer, strip};
      |                                        ^~~~~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:156:47: warning: narrowing conversion of 'strip' from 'int' to 'unsigned int' [-Wnarrowing]
  156 |         const ldmx::HcalID id{section, layer, strip};
      |                                               ^~~~~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:211:57: warning: narrowing conversion of 'id.ldmx::HcalID::section()' from 'int' to 'unsigned int' [-Wnarrowing]
  211 |                       (getScintillatorLength({id.section(), 2, id.strip()}) -
      |                                               ~~~~~~~~~~^~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:211:72: warning: narrowing conversion of 'id.ldmx::HcalID::strip()' from 'int' to 'unsigned int' [-Wnarrowing]
  211 |                       (getScintillatorLength({id.section(), 2, id.strip()}) -
      |                                                                ~~~~~~~~^~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:238:57: warning: narrowing conversion of 'id.ldmx::HcalID::section()' from 'int' to 'unsigned int' [-Wnarrowing]
  238 |                       (getScintillatorLength({id.section(), 2, id.strip()}) -
      |                                               ~~~~~~~~~~^~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/DetDescr/src/DetDescr/HcalGeometry.cxx:238:72: warning: narrowing conversion of 'id.ldmx::HcalID::strip()' from 'int' to 'unsigned int' [-Wnarrowing]
  238 |                       (getScintillatorLength({id.section(), 2, id.strip()}) -
```

and in `HgcrocEmulator.cxx`

```
/Users/tav/Documents/1Research/LDMX/ldmx-sw/Tools/src/Tools/HgcrocEmulator.cxx: In member function 'std::vector<ldmx::HgcrocDigiCollection::Sample> ldmx::HgcrocEmulator::noiseDigi(const int&, const double&) const':
/Users/tav/Documents/1Research/LDMX/ldmx-sw/Tools/src/Tools/HgcrocEmulator.cxx:235:17: warning: narrowing conversion of 'pedestal' from 'double' to 'int' [-Wnarrowing]
  235 |     int adc_tm1{pedestal};
      |                 ^~~~~~~~
/Users/tav/Documents/1Research/LDMX/ldmx-sw/Tools/src/Tools/HgcrocEmulator.cxx:240:24: warning: narrowing conversion of '(pedestal + (((const ldmx::HgcrocEmulator*)this)->ldmx::HgcrocEmulator::noise((* & channel)) / gain)' from 'double' to 'int' [-Wnarrowing]
  240 |     int adc_t{pedestal + noise(channel)/gain};
      |               ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
```

I believe this second one is responsible for the msg @bryngemark referred too.

**To Reproduce**

Just compiling with the latest gcc version

tvami, 2023-09-14:
The first one seems easy to fix, the second one OTOH seems like an intentional narrowing. If the experts confirm that's fine maybe we could just do a `static_cast`? I'm happy to submit a PR if that solution is acceptable.

EinarElen, 2023-09-14:
Making ldmx-sw compile cleanly is something that we should have as a goal, so I'm all for patching this

EinarElen, 2023-09-14:
OTOH i don't think that these would be responsible for the message that @bryngemark referenced. This is a compile time warning, while the one they referenced is a runtime issue

tomeichlersmith, 2023-09-14:
You are correct in assuming the `HgcrocEmulator` narrowing is intentional and so a `static_cast` is appropriate.

I think a quick patch of these would be delightful, but could you hold off until #1198 is merged? I basically want to use a simple PR as a way to check the updates I'm making to the PR Validation infrastructure. Most notably, I'm worried about https://github.com/LDMX-Software/Ecal/issues/29 - a warning message from TDecompSVD heavily pollutes the logs and I'm unsure if the number of warnings it prints is consistent.

EinarElen, 2023-09-14:
I've identified the source of the "TypeError" message. It comes from the hardcoded hcal conditions. 

```
HcalTrigPrimConditionsHardcode=SimpleCSVIntegerTableProvider("HcalTrigPrimDigiConditions",["ADC_PEDESTAL","ADC_THRESHOLD","TOT_PEDESTAL","TOT_THRESHOLD","TOT_GAIN"])
HcalTrigPrimConditionsHardcode.validForAllRows([ 1 , # ADC_PEDESTAL -- should match value from HgcrocEmulator
                                                 5 , # ADC_THRESHOLD -- current noise is 
                                                 1,  # TOT_PEDESTAL -- currently set to match ADC pedestal
                                                 10000,  # TOT_THRESHOLD -- rather large value...
                                                 2.5 ] # TOT_GAIN, ratio of recon TOT gain over recon ADC gain
                                               )
```
This is a table of integers but `TOT_GAIN` is set to 2.5

tomeichlersmith, 2023-09-15:
Alright, #1200 showed that the logs were being updated when the "gold" histograms are updated as well so we are good for new PRs. I also had an update I wanted to make #1201 and so no rush on this if other things come up.

tvami, 2024-08-31:
I found one more of this type
```
/sdf/home/t/tamasvami/BugFixes/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:182:56: warning: narrowing conversion of '(int)(((unsigned char)((int)(flags >> ((int)trigscint::QIEStream::CID_SKIP_POS)))) & ((unsigned char)((int)mask8<1>::m)))' from 'int' to 'bool' [-Wnarrowing]
  182 |   bool isCIDskipped{(flags >> QIEStream::CID_SKIP_POS) &
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  183 |                     mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
/sdf/home/t/tamasvami/BugFixes/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:184:57: warning: narrowing conversion of '(int)(((unsigned char)((int)(flags >> ((int)trigscint::QIEStream::CID_UNSYNC_POS)))) & ((unsigned char)((int)mask8<1>::m)))' from 'int' to 'bool' [-Wnarrowing]
  184 |   bool isCIDunsync{(flags >> QIEStream::CID_UNSYNC_POS) &
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  185 |                    mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~   
/sdf/home/t/tamasvami/BugFixes/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:186:59: warning: narrowing conversion of '(int)(((unsigned char)((int)(flags >> ((int)trigscint::QIEStream::CRC1_ERR_POS)))) & ((unsigned char)((int)mask8<1>::m)))' from 'int' to 'bool' [-Wnarrowing]
  186 |   bool isCRC1malformed{(flags >> QIEStream::CRC1_ERR_POS) &
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  187 |                        mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
/sdf/home/t/tamasvami/BugFixes/ldmx-sw/TrigScint/src/TrigScint/QIEDecoder.cxx:188:59: warning: narrowing conversion of '((((int)flags) >> ((int)trigscint::QIEStream::CRC0_ERR_POS)) & ((int)mask8<1>::m))' from 'int' to 'bool' [-Wnarrowing]
  188 |   bool isCRC0malformed{(flags >> QIEStream::CRC0_ERR_POS) &
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  189 |                        mask8<QIEStream::FLAG_SIZE_BITS>::m};
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
```

Not sure what to do with it tho, @bryngemark any suggestions on how this was meant to be? Should we just `static_cast` it?

closed 2024-09-05 by tvami