LDMX-Software/ldmx-sw #1228: Set up signal generation in LYSO target

bryngemark, 2023-11-15:
**Is your feature request related to a problem? Please describe.**
See issue #1225. 

**Describe the solution you'd like**
We need MadGraph libraries + any additional interfacing or patching required to generate signal samples using a LYSO target. LYSO is a composite material which may or may not require a few tweaks to our method.

tomeichlersmith, 2023-11-15:
On the MadGraph library generation side, the main tweaks will be making sure the MadGraph setup can generate libraries for the required elements within LYSO. I've made an issue for this https://github.com/LDMX-Software/dark-brem-lib-gen/issues/16

On the G4DarkBreM side, not much (if anything) needs to change. As I mentioned in the meeting (but I'm repeating here for future reference), G4DarkBreM has already been developed and tested with composite materials so we can use it when generating signal within the ECal for the EaT analysis channel. There, it was able to handle simulating dark brem within PCB, pure W, pure Si, a carbon-air mixture, and a glue mixture. I think LYSO would be able to be handled similarly. In fact, the distributions that G4DarkBreM samples from change slowly as Z changes, and so we don't even include all of the different elements in those materials. We just include the four mostly likely to undergo dark brem and leave the last two to be simulated "well enough" by sampling from the element closest in Z to it. "Close in Z" amounts to O(10 au) and so (for example), Lu (with Z=71) could be pretty faithfully simulated with a library generated from W (with Z=74).

bryngemark, 2024-01-25:
hi @tvami, great, feel free to dig in! nobody is working on it as far as I know (and they have had plenty of time to let us know in case they were).

closed 2024-04-22 by tvami