LDMX-Software/ldmx-sw #1229: Parameters hardwired for 4 GeV (or 8 GeV) should be configurable/deducible

bryngemark, 2023-11-15:
**Is your feature request related to a problem? Please describe.**
Whenever we change the beam energy, a whole set of filter and biasing thresholds need updating. This is a lot of work going deep into the simulation workings of ldmx-sw, which is not very user friendly, and which is error prone. If we invest some effort now when we are anyway identifying and changing all these settings (as we move defaults from 4 to 8 GeV), we might be grateful later.

**Describe the solution you'd like**
When we have chosen some beam energy, any beam energy dependent setting should automatically change. Since most of our thresholds etc are using fractions of the beam energy, this might as well be how they are defined in the configs. (One exception is trigger thresholds which are studied and tuned, but ironically, there the possibility of using thresholds that are a function of the beam energy is already implemented. Using non-default thresholds is also easily set up as it's a parameter directly in the trigger processor itself, and not buried in calls to anything deeper.) 

**Describe alternatives you've considered**
Another option is to have a different version of each filter, biasing etc, just like we have different functions prepared for different particle guns and whatnot. This looks like an unnecessary multiplication to me.

bryngemark, 2023-11-15:
@jpasc27 if you have an idea of where to start making these changes, and are willing to have a go, please feel free to pick this one up! If not, just listing where you have found necessary changes would be very helpful too.

@tomeichlersmith since you had some examples of how to go about this, please add them here :)

tomeichlersmith, 2023-11-15:
The examples I've implemented are all in the `eat.py` module and I think the best example is the one for `midshower_nuclear`, below is the relevant code where I configure the various simulation user actions that do the filtering.

https://github.com/LDMX-Software/ldmx-sw/blob/3c1058e3a2fdbf0826fb61c61a3d028fee3f748b/Biasing/python/eat.py#L83-L90

I keep some of the more relevant actions to have their parameters given to the function as arguments (`bias_threshold` and `min_nuc_energy`), but the other threshold simply scales with the energy of the passed generator (For EaT, I require the primary electron to reach the ECal with 87.5% of its original beam energy.) The one complexity is that the `ParticleGun` stores its beam energy in GeV whereas most (if not all?) filters use MeV so there is an extra factor of 1000.

The good news is that most[^1] of the energy thresholds within our filters and biasing were written as configurable on the C++ side and so updating them to scale with the beam energy is mostly a case of just updating the python configuration modules i.e. adopting a similar strategy of the EaT case where the thresholds that are "easy" to determine as a fraction of the beam energy are set as a fraction of the beam energy while the rest are left as arguments to the function configuring the simulation.

[^1]: I say "most" because I like to hedge my bets, but in fact I have yet to find any energy threshold in ldmx-sw that is hardcoded in the C++. There are some coded into the python that are more complicated to change, but they are still available to change within the python.

bryngemark, 2023-12-11:
@jpasc27 would you be willing to take on making the modifications to filters you made "on the fly" in your [ecalPN config ](https://github.com/LDMX-Software/LDCS/blob/main/exampleConfigs/ldmxjob_ecalPN.py), but in the configs of each filter/biasing module directly? Since you have the plots from your on-the-fly way of doing it, you would be perfectly positioned to checking that you get the exact same distributions by making the changes deeper down instead. I think most of what you changed (except the biasing factor) can be rescaled by a factor `beamEnergy/4000`.

jpasc27, 2023-12-12:
Yes I believe I can do that! I will make the changes and look at the plots this week and get back to you.

Should I just leave the biasing factor "as is" and change it in the ecalPN config, or would we consider making a new version of it for 8 GeV?

bryngemark, 2023-12-13:
I believe the general idea is that we are moving to 8 GeV as default. So I suggest we update to 550 since a) we can lower to 450 on the fly and b) too high biasing triggers warnings while too low silently causes simulation inefficiency.

tvami, 2024-09-21:
Hey @jpasc27 what's missing for this issue? I only see the trigger update connected to this, was there anything else? And is there anything else to be updated?

tvami, 2025-02-14:
I think https://github.com/LDMX-Software/ldmx-sw/pull/1548 resolved this

closed 2025-02-14 by tvami