LDMX-Software/ldmx-sw #1233: Many test configs have outdated configurations

bryngemark, 2023-11-21:
**Describe the bug**
Not all of the tests can be run. 

**To Reproduce**
```
cd $LDMX_BASE/ldmx-sw
bash-5.2$ for c in `ls */test/*.py`; do echo -e "\n --------------- \nTesting with $c " ; echo ; ldmx fire $c ; done

```
and the full results list is as follows

```
Testing with Biasing/test/ecal_db.py
TypeError: dark_brem() missing 1 required positional argument: 'generator'

Testing with Biasing/test/ecal_pn.py
[ Simulator ] : Started 80917 events to produce 1000 events.
---- LDMXSW: Event processing complete  --------

 Testing with Biasing/test/energy_sort.py
[ Simulator ] : Started 10 events to produce 10 events.
---- LDMXSW: Event processing complete  --------

Testing with Biasing/test/target_db.py
[...]
[ ParallelWorld ]: Adding : sp_hcal_top_bot_PV to parallel world.
[ ParallelWorld ]: Adding : sp_hcal_left_right_PV to parallel world.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Malformed row in CSV file: not exactly 9 columns

 ---------------
Testing with Biasing/test/target_en.py

---- LDMXSW: Loading configuration --------
---- LDMXSW: Configuration load complete  --------
---- LDMXSW: Starting event processing --------

[ ParallelWorld ]: Adding : sp_hcal_top_bot_PV to parallel world.
[ ParallelWorld ]: Adding : sp_hcal_left_right_PV to parallel world.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Malformed row in CSV file: not exactly 9 columns

Testing with Biasing/test/target_en.py
[ Simulator ] : Started 973923 events to produce 53 events.
---- LDMXSW: Event processing complete  --------

Testing with Biasing/test/target_mumu.py
[ Simulator ] : Started 864033 events to produce 264 events.
---- LDMXSW: Event processing complete  --------

Testing with Ecal/test/ecal_digi_pipeline.py
---- LDMXSW: Loading configuration --------
[ WARN ] : Not defining SensDet in GDML since v1.0 of SimCore. See https://github.com/LDMX-Software/SimCore/issues/39 
[...]
[ fire ] 4 : [ConditionUnavailable] : No provider is available for : HcalGeometry

 ---------------
Testing with Ecal/test/ecal_digi_pipeline_test_config.py

---- LDMXSW: Loading configuration --------
Configuration Error [UnableToCreate] : Unable to create instance 'fakeSimHits' of class 'ecal::test::EcalFakeSimHits'. Did you load the library that this class is \
apart of?

 ---------------
Testing with Framework/test/config_python_test_config.py

---- LDMXSW: Loading configuration --------
Configuration Error [UnableToCreate] : Unable to create instance 'test_instance' of class 'framework::test::TestConfig'. Did you load the library that this class i\
s apart of?

 ---------------
Testing with Hcal/test/hcal_digi_pipeline_test_config.py

---- LDMXSW: Loading configuration --------
TypeError: 'float' object cannot be interpreted as an integer

 ---------------
Testing with Hcal/test/hcal_geometry_test_config.py

---- LDMXSW: Loading configuration --------
TypeError: 'float' object cannot be interpreted as an integer

```

**Desired behavior**
All tests run smoothly 

**Environment:**
Output of `ldmx config`:
``` bash
bash-5.2$ ldmx config
LDMX base directory: /Users/lene/work/ldmx-sandbox
uname: Darwin ip59-58.eduroam-employee.wireless.lu.se 22.6.0 Darwin Kernel Version 22.6.0: Wed Jul  5 22:22:05 PDT 2023; root:xnu-8796.141.3~6/RELEASE_ARM64_T6000 arm64
OSTYPE: darwin22.1.0
Bash version: 5.2.15(1)-release
Display Port: docker.for.mac.host.internal
Container Mounts: /Users/lene/work/ldmx-sandbox
Container Environments: 
Docker Version: Docker version 24.0.6, build ed223bc
Docker Tag: ldmx/local:dockerIss77
template parsing error: template: :1:2: executing "" at <index .RepoDigests 0>: error calling index: reflect: slice index out of range
    Digest: 
```

**Additional context**
Add any other context about the problem here.

tvami, 2025-02-14:
1)
```
 ---------------
Testing with Biasing/test/ecal_db.py
TypeError: dark_brem() missing 1 required positional argument: 'generator'
```
-- > 
This misses the input LHE library, we cant fix it directly, but maybe we should have `eat` as a new CI and have this deleted?

2)
```
 ---------------
Testing with Biasing/test/target_db.py
[...]
[ ParallelWorld ]: Adding : sp_hcal_top_bot_PV to parallel world.
[ ParallelWorld ]: Adding : sp_hcal_left_right_PV to parallel world.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Malformed row in CSV file: not exactly 9 columns
```
-- > 
This misses the input LHE library, but we have this exactly in the signal CI. Maybe we should just remove this file. Or not worry about that it doesn't run.

3)
```
 ---------------
Testing with Ecal/test/ecal_digi_pipeline.py
---- LDMXSW: Loading configuration --------
[ WARN ] : Not defining SensDet in GDML since v1.0 of SimCore. See https://github.com/LDMX-Software/SimCore/issues/39 
[...]
[ fire ] 4 : [ConditionUnavailable] : No provider is available for : HcalGeometry
```
-->
The error is easy to fix, it's about importing the `HcalGeometry`. But this pipeline is ran in all the CI tests, so I'd just remove this too.

4)
```
 ---------------
Testing with Ecal/test/ecal_digi_pipeline_test_config.py

---- LDMXSW: Loading configuration --------
Configuration Error [UnableToCreate] : Unable to create instance 'fakeSimHits' of class 'ecal::test::EcalFakeSimHits'. Did you load the library that this class is \
apart of?
```
--> 
This is now ran in the basic tests CI with `Ecal/test/EcalDigiPipelineTest.cxx`

5)
```
 ---------------
Testing with Framework/test/config_python_test_config.py

---- LDMXSW: Loading configuration --------
Configuration Error [UnableToCreate] : Unable to create instance 'test_instance' of class 'framework::test::TestConfig'. Did you load the library that this class i\
s apart of?
```
-->
Same [Framework/test/ConfigurePythonTest.cxx](https://github.com/LDMX-Software/ldmx-sw/blob/b86029d453d04c9dc7830b5adc2f1e591eeeb9cd/Framework/test/ConfigurePythonTest.cxx#L119)

6)
```
 ---------------
Testing with Hcal/test/hcal_digi_pipeline_test_config.py

---- LDMXSW: Loading configuration --------
TypeError: 'float' object cannot be interpreted as an integer
Configuration Error [UnableToCreate] : Unable to create instance 'fakeSimHits' of class 'hcal::test::HcalFakeSimHits'. Did you load the library that this class is apa\
rt of?
```
--> this is tested in the CI tests

7)
```
 ---------------
Testing with Hcal/test/hcal_geometry_test_config.py

---- LDMXSW: Loading configuration --------
TypeError: 'float' object cannot be interpreted as an integer
```
--> content is tested in the CI

8)
```
 ---------------
Testing with Recon/test/sim_and_reco.py

---- LDMXSW: Loading configuration --------
Traceback (most recent call last):
  File "/Users/lene/work/ldmx-sandbox/ldmx-sw/Recon/test/sim_and_reco.py", line 21, in <module>
    sim.biasingOn(True)
AttributeError: 'simulator' object has no attribute 'biasingOn'
```
--> content is tested in the CI

Altogether I think we should remove all the configs listed above, and add a CI test for the EAT

bryngemark, 2025-02-19:
sw dev consensus: most of these tests aren't really contributing beyond validation. we should rather use this space for _unit tests_. so we suggest 

- delete configs that are already tested in CI suite, and 
- move the rest to a directory `exampleConfigs` since they are good for reference.