LDMX-Software/ldmx-sw #1262: Attempting to view certain branches in a ROOT TBrowser causes a crash of that TBrowser

jpasc27, 2024-02-07:
**Describe the bug**
Viewing my output root file in a TBrowser leads to a crash of the root browser, but only when certain branches (and their subsequent leaves) are selected. The specific branches for my file are listed in the **Additional context** and **Screenshots** sections, along with some warnings printed out to the terminal.

**To Reproduce**
Steps to reproduce the behavior:
1. Download these files: 
[ap_producer.txt](https://github.com/LDMX-Software/ldmx-sw/files/14043348/ap_producer.txt) - change extension to .py once downloaded
[mAp_050_uniform_23_brem.lhe.txt](https://github.com/LDMX-Software/ldmx-sw/files/14056241/mAp_050_uniform_23_brem.lhe.txt) - remove .txt extension once downloaded
[mAp_050_uniform_23_decay.lhe.txt](https://github.com/LDMX-Software/ldmx-sw/files/14056248/mAp_050_uniform_23_decay.lhe.txt) - remove .txt extension once downloaded

2. `ap_producer.py` is the config file. Since I am providing the dark brem and A' decay files, you can just set the run number to match the number on those file names, for clarity. The last value in the command below is the number of events - usually it is higher, but for these purposes a lower number will produce the same problem. Be sure you have sourced the ldmx-sw environment, then type:
`ldmx fire ap_producer.py 23 mAp_050_uniform_23_brem.lhe mAp_050_uniform_23_decay.lhe output.root 1500`

4. Open ROOT and attempt to view your `output.root` file in a TBrowser(). Click on different branches, and within each one, pick a leaf to try and get the respective histogram to show up.

**Desired behavior**
To be able to view every branch/leaf of every class in a ROOT TBrowser() without a crash.

**Screenshots**
The list of available branches in my `output.root` file
[image]

**Environment:**
Output of `ldmx config`:

LDMX base directory: /sfs/qumulo/qhome/jlp4td/LDMX
uname: Linux udc-aw34-13c1 4.18.0-425.10.1.el8_7.x86_64 #1 SMP Thu Jan 12 16:32:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
OSTYPE: linux-gnu
Bash version: 4.4.20(1)-release
Display Port: 
Container Mounts: /sfs/qumulo/qhome/jlp4td/LDMX
Container Environments: 
Singularity Version: 1.2.2
Singularity File: /sfs/qumulo/qhome/jlp4td/LDMX/ldmx_dev_latest.sif

**Additional context**
When selecting the `output.root` file, these warnings appear in the terminal:
```
Warning in <TClass::Init>: no dictionary for class pair<int,ldmx::SimParticle> is available
Warning in <TClass::Init>: no dictionary for class ldmx::SimParticle is available
Warning in <TClass::Init>: no dictionary for class ldmx::SimTrackerHit is available
Warning in <TClass::Init>: no dictionary for class ldmx::SimCalorimeterHit is available
Warning in <TClass::Init>: no dictionary for class ldmx::HgcrocDigiCollection is available
Warning in <TClass::Init>: no dictionary for class ldmx::EcalHit is available
Warning in <TClass::Init>: no dictionary for class ldmx::CalorimeterHit is available
Warning in <TClass::Init>: no dictionary for class ldmx::HcalHit is available
Warning in <TClass::Init>: no dictionary for class ldmx::HcalVetoResult is available
Warning in <TClass::Init>: no dictionary for class ldmx::EventHeader is available
Warning in <TClass::Init>: no dictionary for class ldmx::RunHeader is available
```

**Attempting to view a histogram (leaf) in the SimParticles branch causes the TBrowser() to crash.** However, every other branch appears to be fine, at least for this one file I was looking at.

tomeichlersmith, 2024-02-12:
I want to emphasize that this issue seems to be with _all_ event files output by ldmx-sw. I can observe a crash when trying to view the sim particles branches from a basic inclusive simualtion.
```
ldmx use pro v3.3.3
ldmx fire ${LDMX_BASE}/ldmx-sw/SimCore/test/basic.py
ldmx rootbrowse justSim_10_events.root
<crash>
```

I suspect this is a bug with ROOT 6.22/08 TTree::Draw since that is where the stack trace originates and alternative TBrowsers do not have this issue. e.g. JSROOT:

[image]

## TBrowser Alternatives
- [JSROOT](https://root.cern/js/) No installation necessary since its browser-based. Visit https://root.cern/js/latest/ and click on the the button with the three dots to "upload" a local file for inspection.
- [uproot-browser](https://github.com/scikit-hep/uproot-browser) Python based like uproot so install via pip. Not super well maintained, but you could also just use `uproot` directly.
- [ROOT::RBrowser](https://root.cern/doc/master/classROOT_1_1RBrowser.html) not in our version of ROOT in the container, but if you have access to a newer version of ROOT, might be worth trying out.

tvami, 2024-09-21:
I think our approach to this is to have the ROOT update as discussed in https://github.com/LDMX-Software/docker/issues/102
After that's done, we should come back to this issue.