LDMX-Software/ldmx-sw #1265: Set up a processor to identify the beam electron positions at the target

bryngemark, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
There are many many performance studies that would benefit from truth info on where the beam electron hits the target and what momentum it had. 

**Describe the solution you'd like**
I want a processor that takes the simhit collection made by the existing truth hit processor and figures out what the beam electron info is (by grouping them etc). 

**Describe alternatives you've considered**
Scoring plane info? But that has the same issue, that there are several hits along z per incoming electron, which means grouping is needed. Furthermore, currently pileup is not added to scoring planes, meaning that multielectron studies can't rely on this.

bryngemark, 2024-02-21:
@tomeichlersmith @EinarElen 
This builds with `ldmx make Recon` but not with `ldmx make install`.
The error I see in the latter case is 
```
[ 44%] Linking CXX executable fire
/usr/bin/ld: libFramework.so: undefined reference to `ldmx::BeamElectronTruth::BeamElectronTruth()'
collect2: error: ld returned 1 exit status
make[2]: *** [Framework/CMakeFiles/fire.dir/build.make:134: Framework/fire] Error 1
make[1]: *** [CMakeFiles/Makefile2:1391: Framework/CMakeFiles/fire.dir/all] Error 2
```

I have tried this with fresh `build` and `install` dirs, after running `git clean -xxfd` in `Framework`. 

Is there anything in the class implementation itself that I'm missing, or other clues?

tomeichlersmith, 2024-02-21:
The linking error
```
/usr/bin/ld: libFramework.so: undefined reference to `ldmx::BeamElectronTruth::BeamElectronTruth()'
```
is the key clue here. The linker knows that the constructor exists (it is declared in the header), but it is `undefined` meaning that the linker cannot find the code corresponding to this function.

Declaration:

https://github.com/LDMX-Software/ldmx-sw/blob/5d77d6eac4b2b003ebbf347d339a1f48f1f62e5e/Recon/include/Recon/Event/BeamElectronTruth.h#L29

## Solution
Add a definition for the constructor. If you want to do something more complicated, include a definition for it within the source file.
```cpp
BeamElectronTruth::BeamElectronTruth() {
  // extra initialization stuff here
}
```
If you don't have anything to do, you can tell the compiler to write a simple constructor for you with
```cpp
BeamElectronTruth() = default;
```
in the header where the declaration is now.

### Sidenote
It makes sense that the `ldmx make Recon` completes but the `ldmx make install` doesn't. The `ldmx make Recon` just compiles the Recon module but doesn't attempt to link it to the other modules. From the Recon module's point of view, the Event classes are _external_. They are compiled into the Event library with all the other Event classes and linked with the Framework. This last step, linking with the Framework, is what causes this error during `ldmx make install`.

We could update our CMake infrastructure to compile the Event classes into separate libraries corresponding to the different modules they are a part of. This is sort of a necessary step for https://github.com/LDMX-Software/Framework/issues/48 but it is not a necessary step for any of our work. It would just make the grouping of source files into libraries a bit more understandable.

bryngemark, 2024-02-21:
Gah
Thanks @tomeichlersmith 
@EinarElen pointed this out to me but in more generic terms and I only got as far as defining the destructor, and then somehow I overlooked the constructor.

PR #1266 (2024-03-01): Iss1265
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1265 , with the caveat that I don't have a way to get the particle momentum from the TargetSimHits but this might be added later from some other collection if people want to add this info. So I punt on that. 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments

- [x] I ran my developments and the following shows that they are successful.

I ran the inclusive validation config once with 100 events, then ran it again with overlay generating 30 1e inclusive e [... cut]
closed 2024-03-01 by bryngemark