LDMX-Software/ldmx-sw #1286: Make detector volume checks in DetectorConstruction visible to other classes

tvami, 2024-04-18:
**Is your feature request related to a problem? Please describe.**

The functions in https://github.com/LDMX-Software/SimCore/blob/trunk/src/SimCore/DetectorConstruction.cxx
like `isInEcal`, `isInHcal`, `isInTargetRegion`, etc are under the namespace `logical_volume_tests` that's defined in the cxx file only. That means they are not visible to any other class. These functions could be useful especially when we are doing different biasing.

**Describe the solution you'd like**

Declare them in the header of DetectorConstruction 

**Describe alternatives you've considered**

Put them into their own header/source to separate them from their use in detector construction. 
 --> But I think it's ok to reuse them, like this we'll have a smaller code base

**Additional context**

Came up here https://github.com/LDMX-Software/ldmx-sw/pull/1285#discussion_r1569386510

EinarElen, 2024-04-18:
I would probably favour the second option of having a separate translation unit for tools like this if they are useful elsewhere. I would at least be a bit surprised if I saw the detectorconstruction header includes in other places

tvami, 2025-04-08:
Resolved in https://github.com/LDMX-Software/ldmx-sw/pull/1633

closed 2025-04-08 by tvami