LDMX-Software/ldmx-sw #1287: Update EcalVeto BDT for 8 GeV + change hardcoded energy

tvami, 2024-04-20:
**Is your feature request related to a problem? Please describe.**

Currently the ECal Veto BDT is optimized for 4 GeV.
Also just a note that the energy is hardcoded in the code here:
https://github.com/LDMX-Software/Ecal/blob/trunk/src/Ecal/EcalVetoProcessor.cxx#L276

**Describe the solution you'd like**

After the presentation on Monday we plan to update the ECal Veto BDT to the 8 GeV version.
Also make the energy configurable

**Additional context**

See the talk at the SWAN meeting
https://indico.fnal.gov/event/64332/#18-8-gev-bdt-studies-with-the

tomeichlersmith, 2024-05-06:
I'm guessing this is connected to another issue I transferred over from the Ecal repo: https://github.com/LDMX-Software/ldmx-sw/issues/1305

If not, we can leave that one alone, but if it is then we can close two issues :)

tomeichlersmith, 2024-05-06:
I also want to plug #1306 if you're going to already be touching things.

Might be helpful for leaving the old-BDT implementation (with fewer inputs) in a separate processor (maybe named "GabrielleBDT" instead of SingleElectronTargetBDT as mentioned in the linked issue). This could also resolve @bryngemark 's processing time concerns since we could calculate the shower features in large batch and only do tracking+BDT after-the-fact on trigger skimmed samples or something.

danyi211, 2024-05-06:
> I'm guessing this is connected to another issue I transferred over from the Ecal repo: #1305
> 
> If not, we can leave that one alone, but if it is then we can close two issues :)

Yes, this is the new seg-MIP BDT we plan to implement :)

tvami, 2024-05-06:
OK so today's SWAN meeting I think the agreement was that we'll do 2 PRs:
 * first just update so the physics is good with the new BDT 
 * then make all the changes so that the computational aspect is good (i.e. separate MIP tracking)

Not sure about what to do with Gabrielle BDT, I was thinking of just overwriting the current one, but @tomeichlersmith do you see any reason to keep the old one as a separate producer? 

then make all the changes so that the computational aspect is good

tomeichlersmith, 2024-05-06:
Just for comparison, but it seems like (especially after this SWAN meeting) that everyone will step forward to the new one in which case I'm happy to just overwrite the current one. Folks can run with an previous version of ldmx-sw if they want to get the results of gabrielle.

vdutta, 2024-05-06:
Btw, according to [this list](https://confluence.slac.stanford.edu/display/MME/Summary+of+Detector+Geometry+and+BDT+Versions) the new BDT should be called Humberto (!) but maybe we should update to the [2024 set of hurricane names](https://www.nhc.noaa.gov/aboutnames.shtml) and call it Helene instead :D

closed 2024-08-12 by tvami