LDMX-Software/ldmx-sw #1299: Add an ability to ask for N total events to process in addition to the maximum number of events

EinarElen, 2024-05-03:
This comes from https://github.com/LDMX-Software/LDCS/pull/21 

When you don't know roughly how many events you need to try to achieve a certain number of final events, it would be useful if you could (optionally) specify how many events you want in total, e.g. I want to produce 10000 kaon events regardless of how many tries that would take

tvami, 2024-05-03:
This would be very useful for what I'm doing in 
https://github.com/LDMX-Software/ldmx-sw/pull/1289 

Is there a work-in-progress branch for this? Or will any of you @EinarElen @tomeichlersmith PR this, or should I do it?

tomeichlersmith, 2024-05-03:
There is no branch starting to implement this. I created a branch in Framework but never committed anything, so feel free to dig in!

tvami, 2024-05-06:
> desiredEvents (better name later)

what about `totalEvents`?

tomeichlersmith, 2024-05-06:
Yea, I think that's a little better.

I think I'm just running up to the general issue that there are a ton of different ways to count events and we don't want to write a super long variable name that is actually specific (e.g. `totalEventsToProduceWithoutTryLimit`) so I think we should just pick one (`totalEvents` is good) and I'll come back around to this issue on the PR. Maybe we can add more documentation on these configuration options to the config class in this PR as well.

tvami, 2024-05-06:
We could do `maxEventsNoTry` (almost sounds like Master Yoda). Anyway for now I implemented it with `totalEvents`

closed 2024-05-07 by tvami