LDMX-Software/ldmx-sw #1303: Python Config Validation

tomeichlersmith, 2024-05-03:
**Is your feature request related to a problem? Please describe.**
It is _very_ common to misspell parameters and occasionally have parameters with incorrect types. We can avoid this issue by only allowing the insertion of new parameter names to occur _once_ at class construction.

**Describe the solution you'd like**
I sketched out a solution that will check the existence and type of parameters when they are attempting to be set.
```python
       : class Base :
    ...:     def __init__(self, **kwargs) :
    ...:         self.__dict__ = kwargs
    ...:     def __setattr__(self, n, v) :
    ...:         if n in self.__dict__ :
    ...:             if isinstance(v, type(self.__dict__[n])) :
    ...:                 self.__dict__[n] = v
    ...:             else :
    ...:                 raise AttributeError(f'{n} was given as type {type(v)} but should be {type(self.__dict__[n])}')
    ...:         elif n == '__dict__' :
    ...:             super().__setattr__(n,v)
    ...:         else :
    ...:             raise AttributeError(f'{n} is not a defined parameter.')
```
Then child classes would be defined so that their constructor defines the parameters that the class uses.
```python
In [13]: class Child(Base) :
    ...:     def __init__(self) :
    ...:         super().__init__(one = 1, two = 2)
    ...: 

In [14]: c = Child()

In [15]: c.one
Out[15]: 1

In [16]: c.one = 2

In [17]: c.two
Out[17]: 2

In [18]: c.one
Out[18]: 2

In [19]: c.three
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-19-e851540052ba> in <module>
----> 1 c.three

AttributeError: 'Child' object has no attribute 'three'
```

## Loopholes
You can get around this solution by using the hole created for setting the `__dict__` member in the `__init__` method.
```python
old = c.__dict__
old['three'] = 3
c.__dict__ = old
```
This is helpful to know about in case the user needs to add a parameter that isn't in the python module or misspelled there for some reason. This is also hard enough to do that I don't see someone stumbling upon it.

tomeichlersmith, 2024-08-13:
I think what I want is something to define the boiler plate for me like [dataclasses](https://docs.python.org/3/library/dataclasses.html). It'd be _sweet_ to be able to do something like

in config module
```python
@processor('my::Processor','libMyModule.so') # define class and library here
class MyProcessor:
    # define parameters and their defaults here
    one: int = 1
    two: list[float] = [1.0, 2.0]
```

in config script
```python
from MyModule import MyProcessor
MyProcessor(one = 1.0) # Exception because wrong type
MyProcessor(one = 2) # allowed change of default
p = MyProcessor()
p.one = 2 # change of default
p.three = 3 # exception because non-existent parameter
```

Now how to get to the point of being able to define the `class` decorator `processor`.

tomeichlersmith, 2024-08-16:
Got it. After some hacking away to Chappell Roan, I was able to get something operational. Basically, we just hijack `dataclass` and use its internal mapping of known fields in order to implement a `__setattr__` that validates any input attributes (for existence) and values (for type) before accepting them.

The full test file is copied below including the `unittest` I wrote, but the the nice part is that the following syntax is supported.

```python

@parameter_set
class MyParams:
    foo: str = 'bar'

@processor("hello", "world")
class MyClass:
    one: int = 1
    two: float = 2.0
    name: str = 'foo'
    vec: list = [1, 2, 3]
    vec2d: list = [[0.0, 1.0],[-1.0,0.0]]
    p: MyParams = MyParams()

c = MyClass(one = 2.0) # fails, wrong type
c = MyClass(dne = 'dne') # fails, parameter doesn't exist
c = MyClass()
c.one = 2.0 # fails wrong type
c.dne = 'dne' # fails, parameter doesn't exist
c.vec = [1.0, 2, 3] # fails, entries of list are wrong type
```

<details>

  <summary>Full File with Implementation and Testing</summary>

```python

from dataclasses import dataclass, field

def check_list(l, dimension, entry_type):
    if dimension == 1:
        for e in l:
            if not isinstance(e, entry_type):
                raise TypeError(f'Entry {e} is not of expected type {entry_type}')
    elif dimension == 2:
        for ll in l:
            check_list(ll, 1, entry_type)
    else:
        raise Exception(f'Dimension {dimension} not supported.')

def validate_and_set_attr(self, attr, val):
    if attr not in self.__dataclass_fields__:
        raise KeyError(f'Attribute {attr} not a member of {self.__class__.__name__}')

    expected_type = self.__dataclass_fields__[attr].type
    if not isinstance(val, expected_type):
        raise TypeError(f'Attribute {attr} should be type {expected_type} instead of type {type(val)}.')

    if isinstance(val, list):
        expected_dimension  = self.__dataclass_fields__[attr].metadata['dimension']
        expected_entry_type = self.__dataclass_fields__[attr].metadata['entry_type']
        check_list(val, expected_dimension, expected_entry_type)

    self.__dict__[attr] = val

class create_list:
    def __init__(self, l):
        self._list = l

    def __call__(self):
        return self._list

def parameter_set(_class = None, *, post_init = None, **required_parameters):
    def _decorator_impl(cls):
[... 160 lines cut ...]
        with self.assertRaises(TypeError):
            c.p.foo = 1

        with self.assertRaises(TypeError):
            c = MyClass(dne = 'does not exist')

        with self.assertRaises(TypeError):
            c = MyClass(p = 1)
    
    def test_meta_errors(self):
        with self.assertRaises(TypeError):
            @parameter_set()
            class BadVec:
                vec: list = [1, 1.0]

        with self.assertRaises(TypeError):
            @parameter_set()
            class Bad2DVec:
                vec: list = [[2, 3], 1]

if __name__ == '__main__':
    unittest.main()
```

</details>

tomeichlersmith, 2024-09-17:
From #1458 I realized that it would be nice to have a method for specifying legacy/deprecated parameter names. This would include

- updating `__setattr__` to forward "old" parameter names to "new" parameter names (including potentially moving to a sub-parameter-set like is done with the logging parameters)
- optionally adding a deprecation warning to be printed if the "old" parameter name is used. This would be helpful in the case where the "old" parameter name will need to be abandoned or changed in meaning.

Idea outline

```python
@parameter_set
class SubParameters:
    param: float = 1.0

@parameter_set
class MyParameters:
    new_param1: float = 1.0
    new_param2: float = 1.0
    sub: SubParameters
    __legacy__ = {
        'old_param1' : 'new_param1', # simple remap, just would change name silently
        'old_param2' : ('new_param1', True), # True signifies that we should add deprecation message
        'old_param3': 'sub.param', # should be able to propagate to sub-parameter-sets by splitting on `.` character
    }
```

Maybe just have a separate dunder class variable called `__deprecate__` instead of having to support the clunky 2-tuple?

tvami, 2024-09-17:
Yes, this seems to be already in a good shape, will you PR it?

tomeichlersmith, 2024-09-17:
I have not gotten around to going through and updating all the downstream python modules which is the main thing preventing me from merging it. It is high on my list since I think parameter mis-spellings is one of the most vexing issues we see on the regular.

tomeichlersmith, 2025-01-22:
I got the `__legacy__` stuff working. I don't think implementing a flag to disable the deprecation is helpful since, in that case, we would be committing to supporting both forks in the long term which I don't think we should do. Now the syntax is

```python
@parameter_set
class SubParameters:
    param: float = 1.0

@parameter_set
class MyParameters:
    new_param1: float = 1.0
    new_param2: float = 1.0
    sub: SubParameters
    __legacy__ = {
        'old_param1' : 'new_param1', # simple remap, just would change name silently
        'old_param2' : ('new_param1', True), # True signifies that we should add deprecation message
        'old_param3': 'sub.param', # should be able to propagate to sub-parameter-sets by splitting on `.` character
    }

p = MyParameters()
p.old_param1 = 2.0 # sets new_param1 to 2.0 after issuing deprecation warning
p.new_param1 = 3.0 # updates new_param1 to 3.0 without warning
p.old_param3 = 4.0 # updates p.sub.param to 4.0 after issuing warning
p.sub.param = 2 # fails, wrong type
p.dne = 'foo' # fails, does not exist
p.sub.dne = 'bar' # fails, does not exist
p.old_param2 = 5 # fails, wrong type after deprecation warning
```

I have still not started a branch of ldmx-sw with these updates because there are over three hundred classes I suspect I will have to update to this new syntax.[^1] We certainly could support both syntaxes (they appear the same on the C++ side), but I think this would also be a good opportunity to refactor the module structure (not just one mega `ldmxcfg.py` file) and potentially add some tests to make sure the core setting/checking/getting loop on the python side is operational.

[^1]: Running `fd -p '.*/python/.*py' --exclude acts -X grep -c class | awk -F: 'BEGIN{s=0} {s += $2} END {print s}'` from the root of the ldmx-sw repo returns 302.

<details>

  <summary>Full Implementation File with Tests</summary>

```python
"""development file for python config parameter sets

Just run this file to run the tests.

    [denv] python3 processor.py -v

The tests are at the bottom and the code that would be
in the configuration pyton module is at the top.
"""

import warnings
from dataclasses import dataclass, field

def check_list(l, dimension, entry_type):
    if dimension == 1:
        for e in l:
            if not isinstance(e, entry_type):
                raise TypeError(f'Entry {e} is not of expected type {entry_type}')
    elif dimension == 2:
        for ll in l:
            check_list(ll, 1, entry_type)
    else:
        raise Exception(f'Dimension {dimension} not supported.')

def validate_and_set_attr(self, attr, val):
    # first, we check if the attr is a legacy name
    legacy_remap = getattr(self.__class__, '__legacy__', None)
    if legacy_remap is not None and attr in legacy_remap:
        new_attr = legacy_remap[attr]
        warnings.warn(
            f"Legacy name '{attr}' has been replaced by '{new_attr}'.",
            DeprecationWarning,
            #stacklevel=2
[... 226 lines cut ...]
                vec: list = [1, 1.0]

        with self.assertRaises(TypeError):
            @parameter_set()
            class Bad2DVec:
                vec: list = [[2, 3], 1]

    def test_legacy_remap(self):
        c = MyClass()
        with self.assertWarns(DeprecationWarning):
            c.pfoo = 'baz'
        self.assertEqual(c.p.foo, 'baz')
        with self.assertWarns(DeprecationWarning):
            c.Name = 'NewName'
        self.assertEqual(c.name, 'NewName')
        with self.assertWarns(DeprecationWarning):
            c.className = 'OldClass'
        self.assertEqual(c.class_name, 'OldClass')

if __name__ == '__main__':
    unittest.main()

```

</details>

tvami, 2026-02-09:
@tomeichlersmith 

doing
```python
class EcalDigiProducer(Producer):
    def __init__(self, instance_name='ecalDigis', si_thickness=0.4):
        super().__init__(instance_name, 'ecal::EcalDigiProducer', 'Ecal')
        self.MeV = compute_value(si_thickness)
```
to

```python
@dataclass
class EcalDigiProducer(Producer):
    si_thickness: float = 0.4
    hgcroc: any = field(init=False)
    MeV: float = field(init=False)
    
    def __post_init__(self):
        self.hgcroc = EcalHgcrocEmulator()
        self.MeV = compute_value(self.si_thickness)
```

is the correct thing, right?

tvami, 2026-02-13:
> `test_dict: MyParameters = ldmxcfg.field(default_factory = MyParameters)`

So there is no way to do something like 
```
 test_dict: list[str,float] =  [ [ one, two], [1, 2.0] ]
```

OK I guess by the time I wrote this down I understood that the issue is that one is a `float` and the other is an `int`. But assuming they'd both be `float` s: one could do this, right?