LDMX-Software/ldmx-sw #1306: Break-Up Ecal Veto

tomeichlersmith, 2024-05-03:
Currently, the `EcalVetoProcessor` is very hefty. Moreover, a lot of the variables calculated by the veto processor can be used by other analyses. With this in mind, my proposal is to break up the current veto processor into different processors that create different event bus objects.

1. SingleElectronShowerFeatures - calculate the shower-shape features for the single-electron BDT. These shower-shape features are also helpful for other analyses that assume single electron (e.g. EaT), so we don't want to put them into the BDT quite yet.
2. Tracking - reconstruct straight and linear-regression tracks. These are helpful for other analyses (e.g. EaT), but could also be helpful for multi-electron anaylses.
3. SingleElectronTargetBDT - BDT for separting target-based dark brem from ecal PN. This is to separate it from other ML that could be focused on ecal-based dark brem (EaT) or another ML method (DNN).

Hopefully, breaking up the ecal veto in this way will make it more maintain-able. This will also be cause for some additions to the event model, but I can hold off from removing the `EcalVetoResult` object until we are comfortable with breaking on-disk backwards compatibility.

danyi211, 2024-05-29:
@tomeichlersmith @bryngemark The new version of BDT will include the number of straight tracks from MIP tracking in the feature list. It might be tricky to separate the MIP tracking out. The only way we could imagine is to train two versions of BDT, one with number of straight tracks and one without. But then the usual way of referring to the number of events after the BDT cut will be ambiguous. The time to evaluate two BDTs might be the same scale as running MIP tracking in the first place.

tomeichlersmith, 2024-05-29:
I'm not saying folks who want to run the BDT should not be required to run MIP tracking, what I'm saying is there are a lot of folks who just use the shower features (like me for instance).

Perhaps the new BDT would require MIP tracking. In this way, the BDT processor would require the MIP Tracking processor to have been run before it. Perhaps another BDT does not require the MIP tracking and as such that other BDT processor would not require the MIP Tracking processor to run before it. Factorizing the code in this way allows for people to opt in for certain requirements if they want. Does that make sense?

bryngemark, 2024-06-05:
as a side note, how is it that the BDT requires MIP tracking nowadays? when it was developed, MIP tracking rejected the last 10 events that the BDT didn't already reject. is it entirely unthinkable to have a separate MIP-tracking based veto step (perhaps in the context of a BDT if that's needed) that is only applied after a first BDT fails to reject an event?

tomeichlersmith, 2024-06-05:
The number of straight and the number of linear-regression MIP tracks are included as feature inputs to the newer BDTs. I think I agree with you that a "fast" BDT would be nice to have since the MIP tracking is time consuming especially on events with a lot of hits that are going to be rejected anyways for other easier reasons.

Another reason (from my point of view) to factorize so that different BDTs and selections can be more explicity about their requirements.

tvami, 2024-06-05:
> how is it that the BDT requires MIP tracking nowadays?

that's the "mip" in "segmip" :P 

I think our plan is to have:
 * package that makes the variables stored in "EcalShowerFeatures"
 * package that makes the MIP tracking variable
 * package that reads in those above and runs the BDT
     - we could experiment what happens if we plug "-1"s in the BDT to the MIP tracking features first, and then have a second  BDT that does the full fledged "segmip"

tvami, 2024-06-05:
(sorry Danyi, I think we typed at the same time! [but at least we are saying the same thing :D ])

bryngemark, 2024-06-06:
thanks all for your patience, I think I was a little sloppy -- I understand and am aware that it _is_ included, I was mostly wondering why it was considered necessary. 

but anyways! it sounds like we have a good path ahead, I think the suggested split sounds really great and will cover the different needs really well.

tvami, 2024-10-14:
I ran a version where MIP tracking is set to -1 for the BDT input in both signal and bkg:
https://github.com/LDMX-Software/ldmx-sw/actions/runs/11320865028
The results are not terrible, but clearly I need a bigger stat study.

Signal:
[EcalVetoResults_EcalVetoResults_bdt_disc.pdf](https://github.com/user-attachments/files/17366582/EcalVetoResults_EcalVetoResults_bdt_disc.pdf)

Kaons:
[EcalVetoResults_EcalVetoResults_bdt_disc.pdf](https://github.com/user-attachments/files/17366588/EcalVetoResults_EcalVetoResults_bdt_disc.pdf)

EcalPN:
[EcalVetoResults_EcalVetoResults_bdt_disc.pdf](https://github.com/user-attachments/files/17366587/EcalVetoResults_EcalVetoResults_bdt_disc.pdf)

tomeichlersmith, 2025-02-20:
I want to push strongly against only pulling out the MIP tracking. The idea behind braking up the processor is not only for performance reasons but also to make the code more maintainable and understandable. If all that can be done at this time is to pull out the tracking, that's okay and definitely a step in the right direction. I just also want to either keep this issue open or open a new one that is still focused on separating the processor into its constituent parts.

tvami, 2025-02-20:
We could try to have more functions it the EcalVeto, but w/o the tracking it's a 500 lines of code so it's not too bad. There is a lot of interdependence in the rest so I'm not fully convinced that having a 3rd producer will help the maintenance / understandability. Form the point of view of the cutflow I'm imagining for the analysis: everything that goes into the BDT should be one producer / veto, and then there is a MIP tracking veto. (With this I'm also saying not to use MIP tracking in the BDT).

Does having more functions in the BDT veto file sound like a compromise to you?

tvami, 2025-02-20:
Idk maybe it's fine to do what you propose, let's come back to it after we took up the MIP tracking

tomeichlersmith, 2025-02-20:
I guess I just see the calculation of these features as naturally disconnected from running a BDT or some other form of ML inference which makes it seem "natural" to me to separate them into different processors. Its not really lines of code to me, its more the purpose of the code.

PR #1738 (2025-07-02): Break apart Mip Tracking from EcalVetoProcessor
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
Resolves #1306 

The linear regression stage of the Mip Variables takes up most of the time in the cutflow, and to save time, we've moved part of this to the new EcalMipProcessor

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments.
- [x] I read, understood and follow the [coding rules](https://ldmx-software.github.io/developing/coding-rules.html).

- [x] I ran my developments and the following shows that they are successful.
closed 2025-07-02 by tvami