LDMX-Software/ldmx-sw #1308: Compiling stand-alone C++ Processors from Python Config

tomeichlersmith, 2024-05-03:
While writing https://ldmx-software.github.io/analysis/ldmx-sw.html I realized that it is relatively easy to write a stand-alone, single-file analyzer (or producer for that matter) due to our library loading mechanism. I also realized that, since the library loading happens _after_ the python config has been fully run, we can do the compilation of the stand-alone single-file processor within the python config.

For this to work, I currently have to write my own `class` that mimics the `LDMX.Framework.ldmxcfg.EventProcessor` class but just omits the `addModule` within the constructor. (This is done in the linked documentation.)

https://github.com/LDMX-Software/ldmx-sw/blob/11d6a8e89250b357885602075c88b423dc04cdeb/Framework/python/ldmxcfg.py.in#L34-L39

If we updated `EventProcessor` to be something like
```python

from pathlib import Path
import subprocess

class EventProcessor:
    def __init__(self, instanceName, className, library):
        self.instanceName = class_name
        self.className = class_name
        self.histograms = []

        if library.hassuffix('.so'):
            # assume user is passing a filepath to load
            Process.addLibrary(library)
        else:
            # assume user has just passed the name of the module the class is compiled into
            Process.addModule(library)

    def from_file(fp, class_name = None, instance_name = None, **config_kwargs):
        if not isinstance(fp, Path):
            fp = Path(fp)
        if not fp.is_file():
            raise ValueError(f'{fp} is not accessible.')
        
        src = fp.resolve()
        if class_name is None:
            # assume class name is name of file if not provided
            class_name = fp.stem
        if instance_name is None:
            # use class name for instance name if not provided
            instance_name = class_name
        
        lib = fp.parent / f'lib{fp.stem}.so'
        if not lib.is_file() or fp.stat().st_mtime > lib.stat().st_mtime:
            print('Analyzer source is newer than the library, recompiling...') # could make this optional
            import subprocess
            subprocess.run([
              'g++', '-fPIC', '-shared', '-o', str(ta_lib.resolve()),
              '-lFramework', '-I/usr/local/include/root', str(ta_src.resolve())
            ], check=True)

        instance = EventProcessor(instance_name, class_name, str(lib.resolve()))
        for cfg_name, cfg_val in config_kwargs:
            setattr(instance, cfg_name, cfg_val)
        return instance
```
Then we could have configs loading single-file processors. For example
```cpp
// in MyAnalyzer.cxx
#include "Framework/EventProcessor.h"

#include "Ecal/Event/EcalHit.h"

class MyAnalyzer : public framework::Analyzer {
 public:
  MyAnalyzer(const std::string& name, framework::Process& p)
    : framework::Analyzer(name, p) {}
  ~MyAnalyzer() = default;
  void onProcessStart() final;
  void analyze(const framework::Event& event) final;
};

void MyAnalyzer::onProcessStart() {
  // this is where we will define the histograms we want to fill
}

void MyAnalyzer::analyze(const framework::Event& event) {
  // this is where we will fill the histograms
}

DECLARE_ANALYZER(MyAnalyzer);
```
could be run with
```python
// config.py
from LDMX.Framework import ldmxcfg
p.sequence = [ ldmxcfg.Analyzer.from_file('MyAnalyzer.cxx') ]
p.inputFiles = [ 'events.root' ]
p.histogramFile = 'hist.root'
```

tvami, 2024-05-03:
I think this would be awesome!

EinarElen, 2024-05-03:
Looks great!

PR #1310 (2024-05-13): Python config cleanup and enable single-file standalone processors
I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?

This resolves #1308 and #1309 . Resolving #1309 was simply a chore, dropping the `.in` suffixes on some of the python files allows for them to be better syntax highlighted in our text editors and on GitHub. These files are still being configured by CMake so when building happens allowing for them to access build information (for example the install prefix), but since we can follow CMake's [configure_file](https://cmake.org/cmake/help/l [... cut]
closed 2024-05-13 by tomeichlersmith