LDMX-Software/ldmx-sw #1312: [Discussion] Make beam-spot smearing the default

tomeichlersmith, 2024-05-06:
This came up in the latest SW Dev meeting while talking about validating the ECal PN sample for 8GeV beam particles. It is a common occurrence (especially for new users) to forget the extra config line that enables the standard LDMX beam spot smearing. Maybe we want to make the standard beam spot smearing the default and then allow users to disable it if they so wish?

This would be a very easy change to make, but it has the potential to break many prior configs that rely on _not_ smearing the beam spot. Calling for all thoughts possible! @bryngemark @EinarElen @tylerhoroho-UVA @jpasc27 

### Proposed Changes
Update default beam-spot smear

https://github.com/LDMX-Software/SimCore/blob/f1988b5579dd00ae27ad5c6f089881d8ee5e6248/python/simulator.py.in#L75

to

```python
        self.beamSpotSmear = [ 20., 80., 0. ]
```

and add a new function to make disabling the beam spot a little more ergonomic.

```python
    def disableBeamSpotSmear(self):
        self.beamSpotSmear = []
```

tvami, 2024-05-06:
Do we really need the `disableBeamSpotSmear` function? People who dont want smearing could just do
```
sim = simulator.simulator("mySim")
sim.beamSpotSmear = [0., 0., 0.]
```
right?

tomeichlersmith, 2024-05-06:
That's true, I'm probably just over thinking how ergonomic it needs to be.

tvami, 2024-05-06:
Yeah I think we should just go ahead with
```
 self.beamSpotSmear = [ 20., 80., 0. ]
```
in the default

bryngemark, 2024-05-06:
ergonomically speaking, how about separating the "beam" use case (beam spot smear enabled by default) from the "gun" use case (no smearing)? i suppose for alignment studies and the like, we could even have a "pencilBeam" function, if this is important... but perhaps that usage is a fringe case with mostly seasoned users.

bryngemark, 2024-05-06:
> Actually I have a question about the numbers, I've been seeing these for 4 GeV and for 8 GeV too, would we expect these to be different for 8 GeV? also tagging @bryngemark

same beam spot size for both beam energies, 20x80 mm

tomeichlersmith, 2024-05-06:
I mean, we could "promote" the beamSpotSmear configuration from the Simulator into the generator. This would then allow users to configure whether smearing occurs on a per-generator level and be a bit more ergonomic in the sense that all of the parameters about how the simulation will be started are contained within the generator(s) provided to the simulator.

This would be an even bigger change but would maybe be more understandable in the long run. Basically, we would just include the following lines in a function in the `PrimaryGenerator` base class that can then call the derived class implementation.

https://github.com/LDMX-Software/ldmx-sw/blob/11d6a8e89250b357885602075c88b423dc04cdeb/SimCore/src/SimCore/G4User/PrimaryGeneratorAction.cxx#L114-L137

tvami, 2024-05-06:
> the beamSpotSmear configuration from the Simulator into the generator.

can you maybe explain what was the incentive to put this into sim in the first place?

tomeichlersmith, 2024-05-06:
Largely historical, we wanted to be able to smear the beam spot regardless of which generator was used and, since the generators did not share a common base class at the time, we implemented it in the primary generator action. Later, we moved the generators to be configured by the python configuration instead of Geant4 messengers when we brought the simulation in as as Framework processor (and not just a separate executable) and the beam spot smear remained in the primary generator action even after giving all of the primary generators a common base class.

I cannot think of a physical reason to keep the beam spot smearing separate from the generators. Perhaps if we wanted correlated smearing between multiple generators? But that could just as easily be implemented by a single generator producing multiple primary particles.

closed 2026-02-11 by tvami