LDMX-Software/ldmx-sw #1315: SimTrackerHit Detector ID's are screwed up

omar-moreno, 2024-05-06:
# Issue

After the update to the new detector ID's, the module ID is now 0 for all layers.  This is causing a mismatch between the detector ID from a hit and those associated with a detector element. 

# Solution

The SD needs to be updated so it sets the correct module ID.  Currently, the module numbers are being calculated from the copy numbers.  Going forward, either a dedicated module parameter needs to be added to the GDML or the calculation needs to be fixed.

tvami, 2024-09-21:
@bloodyyugo have you looked at module IDs recently? Is this still an issue?

tvami, 2026-02-11:
Tagger is still at zero
[image]

Recoil is has a lot of zeros but not all

[image]

tvami, 2026-02-11:
OK so the original issue is resolved, tho both the 5th and 6th recoil layers have 10 sensors (modules) which I'm not really sure to be what the design is, but anyway, closing this issue now

[image]

tvami, 2026-02-11:
(I think it was resolved in https://github.com/LDMX-Software/ldmx-sw/commit/f8a58be42431b753c8e4aa0a079243252c3b9991 )