LDMX-Software/ldmx-sw #1317: Better ECal Sim Hit Merging

tomeichlersmith, 2024-05-06:
Our current "solution" for merging ECal sim hits is complicated _and_ unhelpful. We try to store separate contributors for the sim hit, but these separate contribs also need to be merged in order to save space. This merging machinery is completely based on PDG ID right now, which is unhelpful because it doesn't take time into account whatsoever.

We can re-work how EcalHitIO works after v3.0.0 so that we have access to the sim cal hit event object (which will probably be lightened by this).

## Overall Plan:
- Add a Python configuration class for passing parameters to EcalHitIO
- Lighten SimCalHits by removing contribs machinery (all SimCalHits are post any merging)
- Implement configurable decisions on how to merge sim cal hits in the ecal
    - Merge based on time (e.g. which "bunch" are we in)
    - Merge based on track ID (multiple hits in the same cell from the same particle are merged into one hit)
    - Merge based off of particle flavor (multiple electron hits in the same cell are merged into one hit)
- Set position information to zero for _all_ ECal hits because the ID already stores fine-grained enough geometry information and ROOT will compress these branches.
- Will need to edit Ecal digitizer to match these changes (and work with all options)
- Update OverlayProducer in EventProc to clean it up

Calling in comments from @jmmans and @omar-moreno to add anything that I forgot.

tvami, 2024-09-21:
Is this only relevant to 2e (and higher), or already for 1e? Where are we in that overall plan envisioned above?

EinarElen, 2024-09-23:
I think it is worth keeping in mind here that resimulation allows us to drop a lot of sim truth that isn't strictly required for large samples

tvami, 2024-09-23:
> and it is very common to want to know where certain hits "came from".

sorry I didnt get this: isnt the current approach more inclusive and with that helps us to know the the hit came from?

tomeichlersmith, 2024-09-23:
The current approach _can be_ helpful but I don't think it is necessarily helpful. It lies in this middle ground between a simplified view and a fully-detailed view of what happened where folks can get confused. I would rather have a merging strategy where the simplified view is more obvious about what the hits mean and controllable by the person configuring the simulation and leaving the detailed view to resimulation where we opt-in to save all of the information.

tvami, 2026-02-05:
Edited the title to include HCAL too