LDMX-Software/ldmx-sw #1319: Put logging messages into logger format

tomeichlersmith, 2024-05-06:
This isn't very complicated, just cumbersome.

Now that Framework has the logging enabled and configurable from the python, we start updating classes to use the logger. My plan is to leave "verbosity" settings in each of the classes that have them so that a user can have specific channels be very verbose while keeping others quiet for debugging. But the messages can still be sent through the logger framework.

tvami, 2024-05-06:
Does this mean to have all the `cout`-s replaced with 
```
ldmx_log(info)
```
?

Do you wanna break this up to subdirectory level and divide the work among ourselves?

tvami, 2024-05-06:
ok I see, then there is not much dividing to be done! And I also agree it would be good to have just one logging system and not to use G4cout

tomeichlersmith, 2025-07-11:
For posterity: https://github.com/LDMX-Software/ldmx-sw/pull/1767#issuecomment-3063109013