LDMX-Software/ldmx-sw #1344: Spurious PE entries in Hcal hits at around 817 PE

EinarElen, 2024-05-09:
As noted by @jpasc27 and @mrsolt, we seem to have some odd entries in the HCal rechits with occasional photoelectrons > what you get at max ADC (see [Jessica's talk at the 2023 spring collaboration meeting ](https://indico.fnal.gov/event/58323/contributions/259798/subcontributions/8885/attachments/166149/221105/LDMX%20CM%20Hcal%20Studies%20Update.pdf) slide 5). 

As far as I understood it, Jessica does not take the `maxPE` hits from the veto processor so this should be unrelated to LDMX-Software/Hcal#58. It also looks like the hits always have the same value, which wouldn't make sense if it was LDMX-Software/Hcal#58. Unclear if it is related to LDMX-Software/Hcal#21 

I tested this quickly with a neutron gun while working on the HCal DQM today and can confirm that I'm getting the same thing.

[image]

All of the RecHits included in the figure above are from the Back HCal.

tvami, 2025-01-25:
> I tested this quickly with a neutron gun while working on the HCal DQM today and can confirm that I'm getting the same thing.

@EinarElen can you show what you did to reproduce this? I'm wondering if this issue is gone now with having all the initializations fixed?

tvami, 2026-01-17:
Maybe this is slightly different then the one above, but this is what I see
[image]

I've cornered it to be for the case when we are in ADC mode and the ADC saturates. What do we want to do with this case? Flag the hit to be saturated and move on? Change the definition of the PE? Something else? @bechenard what's your take?

tvami, 2026-01-18:
Yes, I understand that this is a feature of the ADC mode, but I wonder if that means we should be in ToT mode in these cases?

tvami, 2026-01-18:
> (if it had maybe 10x statistics) 

here is the same plot with >1M HCal gun events in logY scale now:

[image]