LDMX-Software/ldmx-sw #1351: Bring back the old simple reconstruction chain

EinarElen, 2024-05-09:
When developing, debugging, or trying to understand reconstructed quantities people often have to work with simulation level hits which are quite different from their reconstructed counterparts. This means that people have to re-create an entire (often worse) reco-chain in their analysis.

Having the old reconstruction tool available as an alternative would solve this problem. It should of course be clearly marked as simplified & not the default use.

tomeichlersmith, 2024-05-09:
Draft solution in submodule: https://github.com/LDMX-Software/Hcal/pull/75

cmantill, 2024-07-15:
hi @EinarElen, thinking about using this for clustering plots in the DR and I want to make sure that the position reconstruction is understood.

Could you provide from your file above, a plot comparing the position reconstruction but split x- and y-oriented layers so that the coordinate is given by the layer/bar measurement only? If you are swamped, I could make that plot too if you point me to the output file.

PR #1591 (2025-02-13): bring back old Hcal digi/reco chain for testing/debug
This is a transfer of https://github.com/LDMX-Software/Hcal/pull/75 into this repository.

I am updating _ldmx-sw_, here are the details.

### What are the issues that this addresses?
This resolves #1351 

## Check List
- [x] I successfully compiled _ldmx-sw_ with my developments
- [x] I ran my developments and the following shows that they are successful. See https://github.com/LDMX-Software/Hcal/pull/75
closed 2025-02-13 by tomeichlersmith