LDMX-Software/ldmx-sw #1361: StatBox default in DQM plots

tvami, 2024-06-04:
**Is your feature request related to a problem? Please describe.**

Is there a reason why the statbox is switched off in default for the DQM histograms? If in TBrowser I click on "Options/Statistics" it does not come back. While if we did the other way around (i.e. have statbox on in default) then I can switch it off easily. 
I can see the argument to not have statbox when we compare plot, but one can easily switch off the statbox in the comparison code. For looking at plots themselves I'd say the statbox add important info.

**Describe the solution you'd like**
Change this to "1"
https://github.com/LDMX-Software/ldmx-sw/blob/trunk/Framework/src/Framework/Histograms.cxx#L20

bryngemark, 2024-06-05:
off the top of my head i can only come up with, wonder if having a statbox would make the CI gold/new histo overlays harder to read (as sometimes, wtih an automated placement, they may obscure the distributions). 

that's not a strong argument for removing this information, though; the real value of the histogram comparisons lies in the KS test. i would rather see that we actually use the Validation module to make to overlays for visual comparison, perhaps with a nice ratio panel too, for Failed/Passed separately, perhaps even importing the failed ones to the PR... doing that (artifact downloading, manipulation and reuploading) within GitHub actions only might be tricky (or something we should escalate to GitHub devs?) but i think it's what's needed to reach the full potential of our PR tests also for the non-expert user.

tvami, 2024-06-05:
OK so you'd be open to have it enabled? What do other people thing? @tomeichlersmith @EinarElen @tylerhoroho-UVA

I agree with your points in the second block, the `TRatioPlot` could be a way to go.

tomeichlersmith, 2024-06-05:
I'm fine with enabling it :+1: it was introduced before my time and I haven't bothered with it.

closed 2024-06-10 by tvami