LDMX-Software/ldmx-sw #1365: Change ACTS to v35

pbutti, 2024-06-11:
Change ACTS version to v35. We are currently using v19, which is starting to be deprecated.

bloodyyugo, 2024-08-26:
This currently builds and runs.  Currently doing comparisons with old acts version.  

The branch is: actsv36.0 and you have to pull the new acts submodule.  I did this:  

git checkout actsv36.0
git pull --recurse-submodules

tvami, 2024-08-30:
I rebased the branch on `trunk` so it picks up the changes in the validation scripts, I hope that's not going to make issues on your end (it really shouldnt as the changes are orthogonal)

bloodyyugo, 2024-08-30:
I've attached v19 vs v36 comparison plots for both tagger and recoil.  These are 4gev electrons (I know, living in the past) spread over the target.  

Things look _mostly_ ok.  In the tagger, v36 finds ~2% more tracks but the number of hits-on-track is smaller; the layers hit plot shows they are lost at high layer number which, for the tagger, means further upstream of target.  I'll look into this.  For the recoil, they are nearly identical.  The momentum may be shifted or have higher low-side tail in v36, but it's a small effect.  

[
[acts-v19-v36-tagger-comparisons-4gev.pdf](https://github.com/user-attachments/files/16818852/acts-v19-v36-tagger-comparisons-4gev.pdf)
[acts-v19-v36-recoil-comparisons-4gev.pdf](https://github.com/user-attachments/files/16818854/acts-v19-v36-recoil-comparisons-4gev.pdf)
](url)

tvami, 2024-08-30:
For 8 GeV, the sigmas seem to go higher and with that the pulls seem to be narrower:
[ACTS-comparrison-8GeV.pdf](https://github.com/user-attachments/files/16821870/ACTS-comparrison-8GeV.pdf)

Individual plots are at https://github.com/LDMX-Software/ldmx-sw/actions/runs/10625204180/artifacts/1872981316

bloodyyugo, 2024-08-30:
Oh, thanks...I forgot to post residuals/pulls.  Yeah, the uncertainties are higher (some of that maybe due to fewer hits on tracks for the tagger at least) and the pulls look significantly narrower and more normal.  The width of these pulls is quite a bit closer to one in v36 vs v19...still a bit broad, but much better.  

 I've included my pull plots for both tagger and recoil.  In the recoil the pulls a) were better to begin with, and b) improved less.  

[acts-v19-v36-Recoil-par-pulls.pdf](https://github.com/user-attachments/files/16822120/acts-v19-v36-Recoil-par-pulls.pdf)
[acts-v19-v36-Tagger-par-pulls.pdf](https://github.com/user-attachments/files/16822121/acts-v19-v36-Tagger-par-pulls.pdf)

> For 8 GeV, the sigmas seem to go higher and with that the pulls seem to be narrower: [ACTS-comparrison-8GeV.pdf](https://github.com/user-attachments/files/16821870/ACTS-comparrison-8GeV.pdf) [...]

tvami, 2024-08-30:
Oh yeah the tagger pulls look much better now! 
Also good to see that 8 GeV and 4 GeV have similar behavior

closed 2024-09-16 by bloodyyugo