LDMX-Software/ldmx-sw #1382: Ratio inset for plots in DQM Validation module

tvami, 2024-08-06:
**Is your feature request related to a problem? Please describe.**

Right now the `Validation` module offers to overlay 1D histograms.

**Describe the solution you'd like**

I believe it would be super useful to automatically generate a ratio plot in the bottom of the overlaid plot to show the differences.

**Describe alternatives you've considered**

One could do this outside of the `Validation` module in a private script, but wouldnt it be easier to do this once and have it in ldmx-sw?

tvami, 2024-08-13:
Talked with Pritam privately and he agreed to have a look at this.

tomeichlersmith, 2024-08-21:
I've done this so many times, here is the matplotlib code to do a raw/ratio plot.
```python
fig, (raw, ratio) = plt.subplots(
    nrows = 2,
    sharex = 'col',
    height_ratios = [2, 1], # height of raw panel relative to ratio panel
    gridspec_kw = dict(
        hspace = 0.05 # separation between panels, can be zero to overlap axis lines
    )
)

raw.hist(..., label='One')
raw.hist(..., label='Two')
raw.set_ylabel('Events')
raw.legend()

ratio.plot(one/two)
ratio.set_ylabel('Ratio')
ratio.set_xlabel('X Label / Units')

fig.savefig('filename.png', bbox_inches='tight')
fig.close() # only helpful in scripts like Validation where we want to cleanup figures that are done
```

tvami, 2024-08-21:
Nice! I think what I'd like is to have the ratio inserted in the bottom of the canvas just below the main plot.
An example:
[image]

tomeichlersmith, 2024-08-21:
That is what the above code does. `subplots` creates two "axes" in a gride within the "figure".

**Vocab Change**

matplotlib | ROOT
---|---
figure | TCanvas
axes | TPad
axis | TAxis

tvami, 2024-08-21:
Ahh perfect, I didnt read it carefully! Will you PR it @tomeichlersmith ?

tomeichlersmith, 2024-08-21:
Got a draft on the attached branch :+1: not sure when I'll get a chance to test it

tvami, 2024-08-21:
I can give a try to testing, do you wanna open a draft PR?

PR #1401 (2024-08-29): Including ratio subplot within differ when comparing
The denominator for the ratio is whatever histogram is first in the list. This could lead to problems in two situations:
1. The first file in found in the directory (or given on the command line) does not have the histogram requested and so the denominator just becomes the next histogram that is found.
2. There are zero values in the denominator. This will lead to annoying divide-by-zero warnings which we can silence if there are a lot of them.

We try to inherit the color from the histogram in the ratio plot so that the legend applies. This [... cut]
closed 2024-08-29 by tomeichlersmith